loco-3d/crocoddyl #1: No package example_robot_data

wxmerkt, 2019-09-02:
I've installed crocoddyl from the robotpkg binaries and it is correctly sourced. The package `robotpkg-example-robot-data` is installed. However, I get:

```
Traceback (most recent call last):
  File "bipedal_walking_from_foot_traj.py", line 6, in <module>
    import example_robot_data
ImportError: No module named example_robot_data
```

cmastalli, 2019-09-02:
I think you should upgrade robot-example-data. Did you do it?

wxmerkt, 2019-09-02:
Yes... I just installed and it's the latest version on the binaries (2.0.0).

Also, the croccodyl version doesn't import any of the action models:

```
Traceback (most recent call last):
  File "jump.py", line 21, in <module>
    from crocoddyl import (ActionModelImpact, ActivationModelInequality, ActivationModelWeightedQuad,
ImportError: cannot import name ActionModelImpact
```

Is it possible that the build server hasn't released the new binaries yet?

cmastalli, 2019-09-02:
@nim65s can tell you more about the binary release. I didn't try it myself yet.

There are few classes that haven't been release yet, e.g.:
 - ActionModelImpact,  ImpactModel3D, ImpactModel6D
 - ActivationModelInequality,
 - SolverFDDP, among others.

We tagged this version 0.9.0 because there is not yet everything there.

cmastalli, 2019-09-02:
Regarding the examples and notebooks, it almost everything working with the exception of:
 - jump.py
 - walk.py
 - salto.py

The problem with these examples is we need to release some of the classes that I mentioned above.

wxmerkt, 2019-09-02:
Similarly @nim65s, the 2.0.0 tag on `example-robot-data` in the binaries does not contain the Python loader either.

wxmerkt, 2019-09-02:
@cmastalli thank you for the update! Looking forward to Guilhem's response and will re-try then.

nim65s, 2019-09-02:
@wxmerkt Python loaders are available in `robotpkg-pyXY-example-robot-data`, depending of your major / minor version of python.

I am fixing the dependency from `robotpkg-pyXY-crocoddyl` to `robotpkg-pyXY-example-robot-data` right now.

cmastalli, 2019-09-02:
I added in the README file instruction of how to install the Python loaders in example-robot-data. This information also appears in the README file of Crocoddyl.

cmastalli, 2019-09-07:
I believe this issue has been solved. If not, feel free to reopen it

closed 2019-09-07 by cmastalli