loco-3d/crocoddyl #3: How to install crocoddyl on Mac

ddliugit, 2019-10-09:
I followed the procedure of instructions in GitHub, but it doesn't work on Mac

cmastalli, 2019-10-09:
Thanks for showing interest in Crocoddyl.

Unfortunately we don't Mac support yet. Nonetheless, it's in our list of priorities.

At the moment, we have very low bandwidth, and this support could take few weeks.

Could you help us to make this support and move things faster? I believe it shouldn't take so much time.

Yours,
Carlos

ddliugit, 2019-11-08:
I will try to do that. Once it is done, I will let you know.

cmastalli, 2019-11-08:
Great! Do not forget to start from the devel branch.

I invite you to read this wiki: https://gepgitlab.laas.fr/loco-3d/crocoddyl/wikis/Developer-Guide

cmastalli, 2019-11-11:
Any updates regarding this topic?

ddliugit, 2019-11-12:
I am working on that, but it doesn't work pretty well. This is the first time I am working on this. 
Do you have some materials or examples to recommend?
I would appreciate that!

ddliugit, 2019-11-12:
For Mac I just want to the install procedure like so:

brew tap gepetto/homebrew-gepetto
brew install crocoddyl

wxmerkt, 2019-11-12:
> For Mac I just want to the install procedure like pinocchio:

This is an actively developed project and providing a `brew` recipe may not be on the immediate timeline/horizon. After installing dependencies you can still run `mkdir build ; cd build; cmake .. ; make`.

jcarpent, 2019-11-12:
I think copy/paste of pinocchio.rb file in https://github.com/Gepetto/homebrew-gepetto/tree/master/Formula and replacing pinocchio by the right names should make the job.

cmastalli, 2019-11-12:
@ddliugit it would be great if you can provide the solution as @jcarpent advised you.

Unfortunately, I don't have bandwidth for working on it on short-term. If you manage to do it, then we'll welcome your PR

ddliugit, 2019-11-12:
Thank you very much @jcarpent .
I will work on that. @cmastalli

cmastalli, 2021-01-29:
Now it is possible to install Crocoddyl on Mac using conda as well.
I will close this issue.

closed 2021-01-29 by cmastalli