loco-3d/crocoddyl #4: cannot import action modellimpact in example

ddliugit, 2019-11-08:
Hi, I try to run example in crocoddyl, but the library seems outdated in python.

Can you check?

No actionmodelimpact model appears.

Traceback (most recent call last):
  File "walk.py", line 5, in <module>
    from crocoddyl import (ActionModelImpact, ActivationModelWeightedQuad, ActuationModelFreeFloating, CallbackDDPLogger,
ImportError: cannot import name ActionModelImpact

cmastalli, 2019-11-08:
Please use the devel branch if you cannot wait for the next release.

cmastalli, 2019-11-11:
I will close this issue since it is solved in devel branch.

Feel free to reopen it if it is needed

closed 2019-11-11 by cmastalli

ddliugit, 2020-01-26:
I checked out devel branch, but it doesn't work!
The following is the log.

~/landing/src/crocoddyl$ python 
Python 2.7.12 (default, Oct  8 2019, 14:14:10) 
[GCC 5.4.0 20160609] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> from crocoddyl import ActionModelImpact
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: cannot import name ActionModelImpact
>>> import crocoddyl
>>>

proyan, 2020-01-26:
It is named `ActionModelImpulseFwdDynamics`
`from crocoddyl import ActionModelImpulseFwdDynamics`

ddliugit, 2020-01-26:
Thanks.

I also met some similar problems:

```
ActivationModelInequality
ActuationModelFreeFloating
CallbackDDPLogger
CallbackDDPVerbose
CostModelCoM
DifferentialActionModelFloatingInContact
StatePinocchio
CostModelImpactCoM
```

>>> import ActivationModelInequality
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named ActivationModelInequality
>>> from crocoddyl import ActivationModelInequality
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: cannot import name ActivationModelInequality
>>> from crocoddyl import ActivationModelWeightedQuad
>>> from crocoddyl import ActuationModelFreeFloating
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: cannot import name ActuationModelFreeFloating
>>> from crocoddyl import CallbackDDPLogger
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: cannot import name CallbackDDPLogger
>>> from crocoddyl import CallbackDDPVerbose
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: cannot import name CallbackDDPVerbose
>>> from crocoddyl import ContactModel6D
>>> from crocoddyl import ContactModelMultiple
>>> from crocoddyl import CostModelCoM
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: cannot import name CostModelCoM
>>> from crocoddyl import CostModelControl
>>> from crocoddyl import CostModelFramePlacement
>>> from crocoddyl import CostModelState
>>> from crocoddyl import CostModelSum
>>> from crocoddyl import DifferentialActionModelFloatingInContact
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: cannot import name DifferentialActionModelFloatingInContact
>>> from crocoddyl import ImpulseModel6D
>>> from crocoddyl import ImpulseModelMultiple
>>> from crocoddyl import IntegratedActionModelEuler
>>> from crocoddyl import ShootingProblem
>>> from crocoddyl import SolverDDP
>>> from crocoddyl import StatePinocchio
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: cannot import name StatePinocchio
>>> from crocoddyl.diagnostic import displayTrajectory
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named diagnostic
>>> from crocoddyl.impact import CostModelImpactCoM
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named impact

ddliugit, 2020-01-26:
It works!
`from crocoddyl import ActionModelImpulseFwdDynamics`

ddliugit, 2020-01-26:
Can you help check above other issue?
Thanks.

cmastalli, 2020-01-27:
The examples are updated with the new API, you need to use them.

ddliugit, 2020-01-27:
do you have an updated example code for jump?

I think I found an old version in the repo.

cmastalli, 2020-01-27:
> do you have an updated example code for twist while jumping?

Unfortunately, I don't have it yet.

However, I just merged now an example with a quadrupedal jump, its name is boxfddp_vs_boxddp.

> I think I found an old version in the repo.

This is a backup file, which is not yet migrated to the new API.
I will migrate later, but this is not top priority now.
You can help me to boost this task, of course. But I suggest you to learn first the new API with the current working examples.

ddliugit, 2020-01-27:
Thanks, I am interested in the biped example. Can I do that instead?

I learned new API with current working examples. It worked fine actually.

I am trying to rewrite the code for talos from the `/examples/npydata/jump.py` using new API. 

For the new API, until now I found two old API `ActivationModelInequality` and `CostModelImpactCoM` cannot be replaced with current C++ API.

Anything I can do to restore that API from current C++?

I am not good at C++ and I use Python a lot.

ddliugit, 2020-01-27:
Is there a way to define properties by querying name like previous method?

impact.costs['track16'].cost.activation = crocoddyl.ActivationModelWeightedQuad(np.matrix(wx**2).T)

cmastalli, 2020-01-28:
> For the new API, until now I found two old API ActivationModelInequality and CostModelImpactCoM cannot be replaced with current C++ API. I cannot find the corresponding ones to replace them.

The `ActivationModelInequality` has been renamed by `ActivationModelQuadraticBarrier`. Note that the API has been improved, you need to check the documentation.

The `CostModelImpactCoM` is not available at the moment.  I will work on it next week, however you can start without it.

> Anything I can do to restore that API from current C++?

No, thank you. We have already started the development of `CostModelImpactCoM`. Furthermore, it requires deeper understanding of c++ implementation

ddliugit, 2020-02-15:
Did you implement `CostModelImpactCoM`,  I still cannot find it in the library.

cmastalli, 2020-02-17:
Yes, it is available now, its name is `CostModelImpulseCoM`