loco-3d/crocoddyl #101: We currently have to manually set the contact in the CostDataForce.

wxmerkt, 2019-11-18:
In GitLab by @proyan on Feb 7, 2019, 11:53

I see one solution to the issue of manually setting the contact in the costdataforce.
We could move the definition of ContactModel inside PinocchioModel. So instead of saving the contactmodel in IAM, we save it in IAM.pinocchioModel.ContactModel and IAM.pinocchioData.ContactData.

It is not dirty, and the dynamics and contacts go hand in hand anyways. Does this solution appeal to you?

wxmerkt, 2019-11-18:
In GitLab by @cmastalli on Feb 7, 2019, 14:04

I would include few thoughts about it:
 - The contact cost information doesn't belong to any robot model (i.e. Pinocchio model). We have the same reasoning for the state cost.
 - Manual setting of each cost models (or its references) is a nice feature imposed by the action model. I don't see a problem here.
 - Pinocchio is our engine, however it would be nice to keep open for other options in future. So our force cost should be agnostic of it.
 - Overwriting the definition of Pinocchio model creates confusion and it's error prone.

wxmerkt, 2019-11-18:
In GitLab by @nmansard on Feb 8, 2019, 09:04

I don't understand. Let's discuss it orally.

wxmerkt, 2019-11-18:
In GitLab by @proyan on Feb 8, 2019, 10:56

changed title from **{-Defining the CostDataForce-}** to **{+We currently have to manually set the contact in the CostDataForce.+}**

wxmerkt, 2019-11-18:
In GitLab by @nmansard on Feb 13, 2019, 10:23

assigned to @rbudhira and unassigned @nmansard

wxmerkt, 2019-11-18:
In GitLab by @proyan on Feb 13, 2019, 13:50

unassigned @rbudhira

wxmerkt, 2019-11-18:
In GitLab by @nmansard on Feb 18, 2019, 13:19

@rbudhira I added something in the impact model which is kind of fixing this issue, although it is quite heavy. Could you check?

wxmerkt, 2019-11-18:
In GitLab by @nmansard on Feb 18, 2019, 13:24

After oral discussion: @proyan is ok with the idea. I will propagate that in the DAMFIC.

wxmerkt, 2019-11-18:
In GitLab by @cmastalli on Feb 27, 2019, 22:18

@nmansard any update? Did you propagate it to DAMFIC?

wxmerkt, 2019-11-18:
In GitLab by @cmastalli on Apr 3, 2019, 14:40

removed milestone

wxmerkt, 2019-11-18:
In GitLab by @cmastalli on Sep 19, 2019, 19:09

mentioned in merge request !220

wxmerkt, 2019-11-18:
In GitLab by @cmastalli on Nov 7, 2019, 10:27

@proyan this is an old thread but still valid

gfadini, 2019-12-17:
Hello, I'm facing some similar issues with a contact definition that I'm using in a friction cone cost that I'm trying to implement on solo. Inside the cost I initialize the contact data ```self.contact_data=self.contact.createData(self.data)``` where ```self.contact``` is a ```crocoddyl.ContactModel3D```. Apparently the force in ```self.contact_data.f.linear``` is always a null vector. Has someone else had similar problems I'd like to know is there a workaround to this issue?

cmastalli, 2019-12-17:
I think you are using an old version of Crocoddyl. I recently mentioned it.

Everything you need is developed in devel branch.

closed 2019-12-17 by cmastalli