loco-3d/crocoddyl #112: NumDiff of cost model

wxmerkt, 2019-11-18:
In GitLab by @nmansard on Feb 13, 2019, 20:45

I found an issue with the current implementation: we cannot have stand-alone num-diff of cost models. It would be good to think a little bit and find a design pattern for num-diff of the costs. 

In the meantime, it is always possible to num-diff the entire DAM (but this compute the num-diff of both dynamics and costs.

wxmerkt, 2019-11-18:
In GitLab by @cmastalli on Feb 15, 2019, 15:16

Could you explain us why we cannot do it?

wxmerkt, 2019-11-18:
In GitLab by @proyan on Feb 15, 2019, 15:29

df_dq and df_dx? :D

wxmerkt, 2019-11-18:
In GitLab by @proyan on Feb 15, 2019, 16:40

For a solution, take the df_dx and run them on the calc of the model. We can create a class with it perhaps, and anyone interested can borrow from this class. In such a case, the user only has to implement the calc function.

wxmerkt, 2019-11-18:
In GitLab by @proyan on Feb 15, 2019, 16:45

or we can add a new function in the base class CostModelPinocchio, calcNumDiff. This way the user can just call calcnumdiff from calcdiff and that should be all.

I prefer this solution. If you can agree, I can implement this.

cmastalli, 2020-03-04:
Solved in #661

closed 2020-03-04 by cmastalli