loco-3d/crocoddyl #147: Asserting examples

wxmerkt, 2019-11-18:
In GitLab by @nmansard on Mar 7, 2019, 16:25

Examples (in example folder) comes with log files in examples/log. It would be nice to check that any new commit keep these log files invariant. As it is likely that such a test would be very sensible, it might be better to add such a validation in the CI as warning (non mandatory test). @nim65s is offering to implement that.

wxmerkt, 2019-11-18:
In GitLab by @nim65s on Mar 26, 2019, 12:19

cf https://gepgitlab.laas.fr/gsaurel/crocoddyl/merge_requests/2

wxmerkt, 2019-11-18:
In GitLab by @cmastalli on Aug 29, 2019, 16:56

@nim65s do we pursue this unit-test? If so, could you let me know what it's missing to finish it?

wxmerkt, 2019-11-18:
In GitLab by @nim65s on Aug 29, 2019, 17:42

The issue was that some non signifiant decimal were different from one computer to another, so either this will change to be more precise with the C++, or either we will have to write a smarter test than just textually compare output files

wxmerkt, 2019-11-18:
In GitLab by @cmastalli on Aug 29, 2019, 17:47

I can tell you that the numerical behaviour have been tiny changed in the c++ implementation. However, I didn't update the log files yet. I want to take a bit of timing inspecting this.

In any case, it could be interesting to test again this now.

cmastalli, 2020-03-05:
This was solved

closed 2020-03-05 by cmastalli