loco-3d/crocoddyl #163: Defining a structure for the wiki

wxmerkt, 2019-11-18:
In GitLab by @cmastalli on Apr 1, 2019, 17:19

I have added few files that define a tentative structure for the wiki. Please see them here:
https://gepgitlab.laas.fr/loco-3d/crocoddyl/wikis/home

The structure has the form:
```
Home
Developer Guide
Frequently Asked Questions 
Action/
        Floating in Contact Systems
        Unconstrained Forward Dynamics
Cost/
        CoM Cost
        Force based Cost 
        Frame base Cost
        State and Control Cost
```

Most of these files are empty, with the exception of `Home` and `Floating in Contact Systems`. @nmansard have a look and let's decide about this structure.

wxmerkt, 2019-11-18:
In GitLab by @nmansard on Apr 3, 2019, 08:57

I would suggest to make "home" a more generic intro, math-free, that maybe lists the philosophy of the lib and gives the references to the available documentations. The current "math" could be renamed "Defining an optimal control problem". 
Did you take inspiration of all the sections in the TSID rationale?

@andreadelprete any thought?

cmastalli, 2020-03-20:
I will close this since it has been zero activity from few months.

Feel free to re-open it!

closed 2020-03-20 by cmastalli