loco-3d/crocoddyl #166: Removing quadprog and multicontact-api as dependencies

wxmerkt, 2019-11-18:
In GitLab by @cmastalli on Apr 4, 2019, 10:44

This requires the following actions:
 - Do not export some Python submodule (`__init__.py`) if there aren't installed these dependencies.
 - Describe in the README file that they're optional dependencies.

wxmerkt, 2019-11-18:
In GitLab by @nim65s on Apr 4, 2019, 15:07

assigned to @nim65s

wxmerkt, 2019-11-18:
In GitLab by @proyan on Apr 4, 2019, 16:23

qpOASES has an LGPL3 license. I'll replace quadprog with qpoases, and that should solve the licensing issue.

wxmerkt, 2019-11-18:
In GitLab by @proyan on Apr 4, 2019, 16:25

Question: @nim65s Does that mean that<br/>
1) I should completely remove qpsolvers.py/quadprogWrapper function, or<br/>
2) I can keep it, and simply make qpoasesWrapper the default solver and the default dependency?

wxmerkt, 2019-11-18:
In GitLab by @nim65s on Apr 5, 2019, 07:54

make `qpOASESWrapper` the default solver should be enough I think. But you will have to put the `from quadprog import solve_qp` inside `quadprogWrapper`, to be sure that we will have nothing to do with quadprog unless someone explicitely use `quadprogWrapper`

wxmerkt, 2019-11-18:
In GitLab by @cmastalli on Nov 7, 2019, 10:30

changed title from **{-Making quadprog and multicontact-api as optional-} dependencies** to **{+Removing quadprog and multicontact-api as+} dependencies**

wxmerkt, 2019-11-18:
In GitLab by @cmastalli on Nov 7, 2019, 10:31

In the c++ implementation we have removed these dependencies, and when we have made the full transition (missing c++ unittest) we will remove this dependencies.

cmastalli, 2021-01-29:
This was solved as I removed the legacy Python unit tests

closed 2021-01-29 by cmastalli