loco-3d/crocoddyl #246: FDDP solver unit-test

wxmerkt, 2019-11-18:
In GitLab by @cmastalli on Sep 7, 2019, 13:35

null

wxmerkt, 2019-11-18:
In GitLab by @proyan on Sep 20, 2019, 15:50

assigned to @proyan

wxmerkt, 2019-11-18:
In GitLab by @cmastalli on Sep 20, 2019, 17:55

We don't have the KKT solver in c++. @hammoudbilal will delivering soon.

wxmerkt, 2019-11-18:
In GitLab by @cmastalli on Sep 25, 2019, 18:07

@proyan I have included a unit-test against Python code (https://gepgitlab.laas.fr/loco-3d/crocoddyl/blob/devel/unittest/bindings/test_solvers.py#L127). We just need to add the c++ one.

I also fixed a bug inside the FDDP.

Everything is on devel branch.

wxmerkt, 2019-11-18:
In GitLab by @proyan on Sep 25, 2019, 18:31

Ok. I'll work on it Friday.

wxmerkt, 2019-11-18:
In GitLab by @cmastalli on Sep 25, 2019, 18:33

This depends on #182. Alternatively, you can solve #250.

wxmerkt, 2019-11-18:
In GitLab by @cmastalli on Oct 4, 2019, 13:07

unassigned @proyan

cmastalli, 2019-11-30:
This issue it was solved before !224

closed 2019-11-30 by cmastalli