loco-3d/crocoddyl #255: Build with multi-threads OFF and vectorisation ON results in double free corruption

wxmerkt, 2019-11-18:
In GitLab by @wxmerkt on Sep 20, 2019, 12:21

(On the latest devel branch)

When building with multithreads OFF and vectorisation ON, the benchmarks segfault with a double free corruption.

The following is tested by running talos_legs:

*  Multithreads OFF, vectorisation OFF: OK (8.89 ms)
*  Multithreads ON, vectorisation OFF: OK (5.05 ms)
*  Multithreads ON, vectorisation ON: OK (5.13 ms)
*  **Multithreads OFF, vectorisation ON: double free or corruption (out)**

This is the backtrace for a crash:
```
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff7a24801 in __GI_abort () at abort.c:79
#2  0x00007ffff7a6d897 in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7ffff7b9ab9a "%s\n")
    at ../sysdeps/posix/libc_fatal.c:181
#3  0x00007ffff7a7490a in malloc_printerr (str=str@entry=0x7ffff7b9c870 "double free or corruption (out)")
    at malloc.c:5350
#4  0x00007ffff7a7be75 in _int_free (have_lock=0, p=0x555555ba2a70, av=0x7ffff7dcfc40 <main_arena>) at malloc.c:4278
#5  __GI___libc_free (mem=0x555555ba2a80) at malloc.c:3124
#6  0x00007fffdbe32543 in boost::python::objects::caller_py_function_impl<boost::python::detail::caller<boost::python::detail::member<Eigen::Matrix<double, -1, 1, 0, -1, 1>, pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl> >, boost::python::return_value_policy<boost::python::return_by_value, boost::python::default_call_policies>, boost::mpl::vector3<void, pinocchio::ModelTpl<double, 0, pinocchio::JointCollectionDefaultTpl>&, Eigen::Matrix<double, -1, 1, 0, -1, 1> const&> > >::operator()(_object*, _object*) ()
   from /opt/openrobots/lib/python2.7/site-packages/pinocchio/libpinocchio_pywrap.so
#7  0x00007fffded76e55 in boost::python::objects::function::call(_object*, _object*) const ()
   from /usr/lib/x86_64-linux-gnu/libboost_python-py27.so.1.65.1
#8  0x00007fffded77008 in ?? () from /usr/lib/x86_64-linux-gnu/libboost_python-py27.so.1.65.1
#9  0x00007fffded7dbcb in boost::python::detail::exception_handler::operator()(boost::function0<void> const&) const
    () from /usr/lib/x86_64-linux-gnu/libboost_python-py27.so.1.65.1
#10 0x00007fffdf004493 in boost::detail::function::function_obj_invoker2<boost::_bi::bind_t<bool, boost::python::detail::translate_exception<eigenpy::Exception, void (*)(eigenpy::Exception const&)>, boost::_bi::list3<boost::arg<1>, boost::arg<2>, boost::_bi::value<void (*)(eigenpy::Exception const&)> > >, bool, boost::python::detail::exception_handler const&, boost::function0<void> const&>::invoke(boost::detail::function::function_buffer&, boost::python::detail::exception_handler const&, boost::function0<void> const&) () from /opt/openrobots/lib/libeigenpy.so
#11 0x00007fffded7d96f in boost::python::handle_exception_impl(boost::function0<void>) ()
   from /usr/lib/x86_64-linux-gnu/libboost_python-py27.so.1.65.1
#12 0x00007fffded746c9 in ?? () from /usr/lib/x86_64-linux-gnu/libboost_python-py27.so.1.65.1
#13 0x0000555555631fee in PyObject_CallFunction ()
```

This is on Ubuntu 18.04 using the latest devel branch of Crocoddyl with the latest robotpkg release of Pinocchio.

wxmerkt, 2019-11-18:
In GitLab by @proyan on Sep 20, 2019, 12:49

I see. I don't think that vectorization and multi-threading are tried for 18.04, so I'm not surprised that there is an issue.

Would it be possible for you to try with the debug versions of pinocchio and crocoddyl and post the trace? If not I'll try with the debug version of pinocchio to see where the problem is, currently I can't make out anything from this trace.

wxmerkt, 2019-11-18:
In GitLab by @wxmerkt on Sep 20, 2019, 13:37

Multi-threading is definitely tried for 18.04, I ran all benchmarks for the ICRA submission on 18.04 machines. The problem here may be the vectorisation without the multi-threading. 
Have you tried to reproduce that setting (multithread=OFF, vectorisation=ON) on your machine?

This is built with the debug version of Crocoddyl. I don't have Pinocchio in a local build (but use robotpkg), so won't be able to provide that traceback, I am afraid.

wxmerkt, 2019-11-18:
In GitLab by @cmastalli on Sep 20, 2019, 13:47

@wxmerkt I don't remember exactly, but I found this issue before. Thanks for reporting it, and @proyan thanks for taking care about it.

wxmerkt, 2019-11-18:
In GitLab by @proyan on Sep 20, 2019, 14:34

Ok, I'll try it on 16.04 first, since it is my local machine. If I find this problem, well and good. Else I go on 18.04

wxmerkt, 2019-11-18:
In GitLab by @proyan on Sep 20, 2019, 14:34

assigned to @proyan

wxmerkt, 2019-11-18:
In GitLab by @cmastalli on Oct 13, 2019, 19:15

mentioned in merge request !246

wxmerkt, 2019-11-18:
In GitLab by @proyan on Oct 17, 2019, 14:23

unassigned @proyan

wxmerkt, 2019-11-18:
In GitLab by @cmastalli on Nov 1, 2019, 10:34

@proyan and @wxmerkt do you still have this problem? Could you try it in the latest devel branch with the c++ quadruped benchmark?

wxmerkt, 2019-11-18:
In GitLab by @wxmerkt on Nov 1, 2019, 13:00

I will check and verify again either this weekend or next week.

wxmerkt, 2019-11-18:
In GitLab by @wxmerkt on Nov 5, 2019, 02:40

Yes, it still happens with the latest devel. 

`cmake .. -DENABLE_VECTORIZATION=ON` and then running ./benchmark/quadrupedal-gaits

wxmerkt, 2019-11-18:
In GitLab by @wxmerkt on Nov 5, 2019, 11:47

I have been tracking this down further and fixed a few issues with allocation (WIP) - but it's not fully resolved yet.

I think we missed a few declarations in classes to make them use the aligned allocator. Also I think we have some std::vector / STL containers that do not correctly use the aligned allocator.

E.g.:
http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html

wxmerkt, 2019-11-18:
In GitLab by @jcarpent on Nov 5, 2019, 12:31

Small question before another reply: is this happening only in Python?

wxmerkt, 2019-11-18:
In GitLab by @cmastalli on Nov 5, 2019, 15:01

This benchmark runs c++ and Python code. Is happening in both case?

wxmerkt, 2019-11-18:
In GitLab by @wxmerkt on Nov 5, 2019, 16:42

I am now debugging using the `arm-manipulation` C++ executable (to easily run gdb). It is clear that we have STL containers that do not do alignment correctly.

I have locally fixed frame-placement and an allocation issue in action-base (quasiStatic). It is now failing in the backward pass, the Quuk_[t] multiplication. I have not dug further as the laptop takes several minutes to compile after every change.

wxmerkt, 2019-11-18:
In GitLab by @jcarpent on Nov 5, 2019, 16:49

As I mentioned during the last Crocoddyl call, all the classes in the framework do not align correctly the structure contained inside the std::vector.

Instead of using:
```
std::vector<T>
```
you should use:
```
std::vector<T,Eigen::aligned_allocator<T> >
```
Then Eigen::aligned_allocator will take of calling the aligned allocator for the type T. For all the classes, you should have `EIGEN_MAKE_ALIGNED_OPERATOR_NEW` in the top of the class definition.

wxmerkt, 2019-11-18:
In GitLab by @wxmerkt on Nov 5, 2019, 16:50

Yes, that is exactly what I was just referring to. We do use variable sized but even for the more complex types that contain fixed-sizes (!), there is no aligned_allocator set. This will have to be fixed all across the codebase.

wxmerkt, 2019-11-18:
In GitLab by @jcarpent on Nov 5, 2019, 16:52

Or just call `EIGEN_DEFINE_STL_VECTOR_SPECIALIZATION`

wxmerkt, 2019-11-18:
In GitLab by @jcarpent on Nov 5, 2019, 16:52

If you just want to do only minor modifications to your codebase for quick debugging.

wxmerkt, 2019-11-18:
In GitLab by @jcarpent on Nov 5, 2019, 16:55

But normally, if you all your codebase + dependencies are compiled with the same flags, then everything should work fine.
But if one of your dependency is not built with vectorization, this means that the constructor of the data is not aligned, and you will get corruption of memory.
What are the current dependencies of Crocoddyl that use Eigen?

wxmerkt, 2019-11-18:
In GitLab by @wxmerkt on Nov 5, 2019, 17:08

eigenpy + pinocchio. I can dig deeper into it, but at the moment with only the laptop it takes a long time to compile - so perhaps when I'm back in Edinburgh.

wxmerkt, 2019-11-18:
In GitLab by @cmastalli on Nov 5, 2019, 18:39

We have some contact and impulse classes that uses std::vector from Pinocchio. I believe there is missing some of the macro declaration there.

Is this problem happening in the arm? If so, You should start looking in Contact/Impulse(3D/6D) classes

wxmerkt, 2019-11-18:
In GitLab by @wxmerkt on Nov 6, 2019, 03:04

Sorry - if it is happening in the arm - that should not require contact/impulse?

wxmerkt, 2019-11-18:
In GitLab by @wxmerkt on Nov 7, 2019, 18:43

I have been digging deeper into it:
- One failure is an unaligned memory data structure in `CostDataFramePlacement`. However, `CostDataFramePlacement` has `EIGEN_MAKE_ALIGNED_OPERATOR_NEW`. After changing the `Matrix6` to `Matrix6x` (for rJf), the error gets pushed down:

Notes: eigenpy, pinocchio, crocoddyl have all been built with the `-march=native -mavx` compilation flag.

The new assert is from line 99 (the constructor) of data.hxx in Pinocchio. 

Have you used vectorisation with Pinocchio before @jcarpent ?

wxmerkt, 2019-11-18:
In GitLab by @jcarpent on Nov 7, 2019, 18:54

Yes, I already tried vectorization on Pinocchio.
I found the bug, and it is related to the Tensor class of Pinocchio which is not aligned by default.

I've fixed here: https://github.com/stack-of-tasks/pinocchio/pull/938

wxmerkt, 2019-11-18:
In GitLab by @wxmerkt on Nov 8, 2019, 01:21

@jcarpent Thank you very much for the fix. The issue is still occurring from `data.hxx`

Error:
```
/usr/include/eigen3/Eigen/src/Core/DenseStorage.h:128: Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 32>::plain_array() [with T = double; int Size = 36; int MatrixOrArrayOptions = 0]: Assertion `(internal::UIntPtr(eigen_unaligned_array_assert_workaround_gcc47(array)) & (31)) == 0 && "this assertion is explained here: " "http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html" " **** READ THIS WEB PAGE !!! ****"' failed.
```

Backtrace:
```
#7  Eigen::Matrix<double, 6, 6, 0, 6, 6>::Matrix (this=0x5555557c7810)
    at /usr/include/eigen3/Eigen/src/Core/Matrix.h:259
#8  pinocchio::DataTpl<double, 0, pinocchio::JointCollectionDefaultTpl>::DataTpl (
    this=0x5555557c7530, model=...)
    at include/pinocchio/multibody/data.hxx:99
#9  0x00007ffff75ec048 in crocoddyl::DifferentialActionDataFreeFwdDynamics::DifferentialActionDataFreeFwdDynamics<crocoddyl::DifferentialActionModelFreeFwdDynamics> (this=0x5555557c7450, 
    model=0x555555796560)
    at include/crocoddyl/multibody/actions/free-fwddyn.hpp:52
```

wxmerkt, 2019-11-18:
In GitLab by @wxmerkt on Nov 8, 2019, 08:23

I chased this further and further down the chain but could not fully resolve.

**However:** Following the Eigen documentation, I set my C++ standard to **C++17** for the workspace containing EigenPy, Pinocchio, and Crocoddyl (`-std=c++17`). One compile later... it works out of the box without a single fix. Three days wasted chasing Eigen alignment issues.

wxmerkt, 2019-11-18:
In GitLab by @proyan on Nov 8, 2019, 09:18

@wxmerkt Could you please post the link to the documentation where you found the cure? :)

wxmerkt, 2019-11-18:
In GitLab by @cmastalli on Nov 8, 2019, 12:11

@wxmerkt I just tried different conditions using c++ benchmarks

| benchmark | Vec. ON / Mthread. ON | Vec. OFF / Mthread. ON | Vec. ON / Mthread. OFF |
| ------ | ------ | ------ | ------ |
| lqr        | :ok_hand: | :ok_hand: | :ok_hand: |
| unicycle   | :ok_hand: | :ok_hand: | :ok_hand: |
| arm        |  :x:      | :ok_hand: | :x:       |
| quadruped  | :ok_hand: | :ok_hand: | :ok_hand: |

I am using Pinocchio v2.1.9, Eigen v3.3.7 without changing flags.

wxmerkt, 2019-11-18:
In GitLab by @jcarpent on Nov 8, 2019, 12:49

So, it seems to work on your computer @cmastalli? You still use C++98?

wxmerkt, 2019-11-18:
In GitLab by @cmastalli on Nov 8, 2019, 13:48

Yes, I use c++98.

Note that it is not working for the arm case

wxmerkt, 2019-11-18:
In GitLab by @jcarpent on Nov 8, 2019, 13:57

Is it the same classes behind ARM and others? I would say there is a missing EIGEN_MAKE_ALIGNED_OPERATOR_NEW or a missing std::vector<T,Eigen::AlignedAllocator<T> >

wxmerkt, 2019-11-18:
In GitLab by @cmastalli on Nov 8, 2019, 14:03

Arm and quadruped uses different action models. The former one basically is an `ABA`-based action model, instead the latter one is `forwardDynamics`.

Unfortunately, the `ABA`-based action model doesn't miss the Eigen macro. See here https://gepgitlab.laas.fr/loco-3d/crocoddyl/blob/devel/include/crocoddyl/multibody/actions/free-fwddyn.hpp#L45.

It doesn't have a `std::vector<T,Eigen>` neither, apart of the Pinocchio-data ones.

wxmerkt, 2019-11-18:
In GitLab by @jcarpent on Nov 8, 2019, 14:37

You should, to have the classes properly aligned.

wxmerkt, 2019-11-18:
In GitLab by @jcarpent on Nov 8, 2019, 14:38

By the way, in `DifferentialActionModelFreeFwdDynamics` you should also have EIGEN_MAKE_ALIGNED_OPERATOR_NEW to have Model of Pinocchio aligned in memory too

wxmerkt, 2019-11-18:
In GitLab by @cmastalli on Nov 8, 2019, 14:52

Ok, I didn't know it that this was needed. I will handle it

wxmerkt, 2019-11-18:
In GitLab by @jcarpent on Nov 8, 2019, 14:54

I would suggest to put this field in all your classes IMHO.

wxmerkt, 2019-11-18:
In GitLab by @wxmerkt on Nov 9, 2019, 11:00

I tested with the arm Vec. ON/Multithreading OFF - and it fails as in your case. I made a couple local fixes to some allocations that pushed the crash further down and it was gone when compiled in C++17. It still persisted as a crash in C++17 for Vec. ON/Multithreading ON.

wxmerkt, 2019-11-18:
In GitLab by @wxmerkt on Nov 9, 2019, 11:01

@proyan It is the first option mentioned in http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html

wxmerkt, 2019-11-18:
In GitLab by @cmastalli on Nov 9, 2019, 15:05

@wxmerkt I am confuse, is it's working all the cases with c++17 with only vectorization?

Is arm with c++17 and multithreading the only that doesn't work?

wxmerkt, 2019-11-18:
In GitLab by @wxmerkt on Nov 10, 2019, 16:24

> @wxmerkt I am confuse, is it's working all the cases with c++17 with only vectorization?

I have only tested the `arm-manipulation` on the laptop as I considered it the simplest. Recompilation took ~20 minutes each so I didn't run all the other scripts and tried to find root causes. I agree that running all cases would have been better.

wxmerkt, 2019-11-18:
In GitLab by @jcarpent on Nov 10, 2019, 16:26

@wxmerkt Do you use ccache?

wxmerkt, 2019-11-18:
In GitLab by @wxmerkt on Nov 10, 2019, 16:33

No, I don't in this case - especially when switching CMAKE_BUILD_TYPE and other flags

wxmerkt, 2019-11-18:
In GitLab by @jcarpent on Nov 10, 2019, 16:39

new versions of ccache are smart enough to detect change in the optimization variable and recover from it. At least, this is what I'm observing on my mac ;)

wxmerkt, 2019-11-18:
In GitLab by @jcarpent on Nov 11, 2019, 08:53

@wxmerkt Could you provide me or extract a minimal example of vectorization breack when using Pinocchio?
I will try to debug it.

cmastalli, 2020-05-04:
It seems I found the explanation of this issue.
The problem happens during the creation of data since `make_shared` does not honor requested alignment. For solving this, we need to use the `allocate_shared` thing.

I am working in providing a fix to this issue.

cmastalli, 2020-05-05:
solved by #768

closed 2020-05-05 by cmastalli