loco-3d/crocoddyl #265: arm code generation + benchmark

wxmerkt, 2019-11-18:
In GitLab by @cmastalli on Oct 4, 2019, 18:08

null

wxmerkt, 2019-11-18:
In GitLab by @cmastalli on Oct 4, 2019, 18:08

assigned to @cmastalli

wxmerkt, 2019-11-18:
In GitLab by @cmastalli on Nov 12, 2019, 16:44

unassigned @cmastalli

cmastalli, 2020-03-05:
@proyan could you handle this task?
I would be good to have profiles for the quadruped

wxmerkt, 2020-03-20:
Is this complete or just closed as wontdo?

cmastalli, 2020-03-20:
I do not see any activity, and I won't do it (overbooked).

@proyan are you planning to create some benchmarks with CppAD?

wxmerkt, 2020-03-20:
Ok, thanks, good to have this noted done as it helps to follow along whether an issue was addressed or closed as wontdo.

cmastalli, 2020-03-20:
I tagged with #wontfix, is this sounds good for you?

proyan, 2020-03-20:
Yes, I'm creating a benchmark with cppAD. I'm using the arm as an example.

proyan, 2020-03-20:
Currently there is an issue with using pinocchio with code-gen. I'll make a PR next week for that. I've been quite busy with another task, so code-gen has gone back to my to-do list.

cmastalli, 2020-03-20:
@proyan thank for pushing this task.

Could you let me know (or pointing out) the issue of Pinocchio?

cmastalli, 2020-05-01:
Solved by #758

closed 2020-05-01 by cmastalli