loco-3d/crocoddyl #278: Add proper virtual keyword

wxmerkt, 2019-11-18:
In GitLab by @wxmerkt on Oct 23, 2019, 12:45

Currently, the implementations of (pure) virtual functions do not carry the `virtual` keyword. This is prone to introduce bugs and also requires copy-paste/duplication in the bindings. We should add the `virtual` (C++98) or `override/final` (C++11, once switch confirmed) keywords.

wxmerkt, 2020-01-17:
This just came back to bite me badly with overrides in Python and C++. We should make fixing the polymorphism and cleaning up the API a priority. There's a lot of duplication that can be removed with proper inheritance.

cmastalli, 2020-01-17:
> This just came back to bite me badly with overrides in Python and C++. We should make fixing the polymorphism and cleaning up the API a priority. There's a lot of duplication that can be removed with proper inheritance.

Could you handle this task?
Unfortunately, I don't have any bandwidth to work on it before beginning of February.

wxmerkt, 2020-01-17:
I won't have bandwidth before March. New action + contact model takes priority for me.

closed 2020-03-13 by cmastalli