loco-3d/crocoddyl #279: Generate and install CMake config

wxmerkt, 2019-11-18:
In GitLab by @wxmerkt on Oct 23, 2019, 13:34

Analot to https://github.com/stack-of-tasks/eigenpy/issues/98, https://github.com/stack-of-tasks/pinocchio/issues/914

wxmerkt, 2019-11-18:
In GitLab by @cmastalli on Nov 1, 2019, 10:40

@wxmerkt are you handling this issue?

wxmerkt, 2019-11-18:
In GitLab by @wxmerkt on Nov 1, 2019, 12:59

No, this requires updates and support from jrl/cmake-modules first. I would put this after ~SAB or happy for someone else to add it.

cmastalli, 2020-03-05:
@wxmerkt is there any update regarding this?

I'm not sure what if I understand the issue, could you explain me?

wxmerkt, 2020-03-05:
It's to update the `CMakeLists` setup so that a `crocoddyl-config.cmake` is generated such that users/dependent projects can:

```cmake
find_package(crocoddyl REQUIRED)

add_executable(example example.cpp)
target_link_libraries(example crocoddyl::crocoddyl)
```

No need for PkgConfig and setting CFLAGS_OTHERS etc.

nim65s, 2020-03-06:
Hi,

I'm already doing this on all SoT packages, loco-3d ones will follow.

cmastalli, 2020-04-28:
@nim65s I think you integrated this thing. Can we close this issue?

nim65s, 2020-04-28:
yep

closed 2020-04-28 by nim65s