loco-3d/crocoddyl #291: Currently pinocchio::impulseDynamics allocates internally memory, do not use it for multithreading

wxmerkt, 2019-11-18:
In GitLab by @cmastalli on Nov 9, 2019, 11:14

I have been working on #271, and I noticed that part of the non-deterministic behavior of the solver is that `pinoccho::impulseDynamics` allocates internally memory, and I reported it few weeks ago: https://github.com/stack-of-tasks/pinocchio/issues/881.

@wxmerkt and @proyan be aware of it, and do not waste your time.

For more elaborate discussion, I suggest to use the Pinocchio issue.

wxmerkt, 2019-11-18:
In GitLab by @cmastalli on Nov 9, 2019, 11:15

changed title from **Currently {-impulse d-}ynamics creates interally memory, do not use for multithreading** to **Currently {+pinocchio::impulseD+}ynamics creates interally memory, do not use for multithreading**

wxmerkt, 2019-11-18:
In GitLab by @cmastalli on Nov 9, 2019, 11:16

changed title from **Currently pinocchio::impulseDynamics {-creates interally memory, do not use-} for multithreading** to **Currently pinocchio::impulseDynamics {+allocates internally memory, do not use it+} for multithreading**

wxmerkt, 2019-11-18:
In GitLab by @wxmerkt on Nov 9, 2019, 11:22

Thank you for the note, I wasn't aware of the upstream issue. It did already waste quite a lot of time.

wxmerkt, 2019-11-18:
In GitLab by @proyan on Nov 9, 2019, 18:14

Cool! Good work Carlos

cmastalli, 2020-04-30:
solved in #765

closed 2020-04-30 by cmastalli