loco-3d/crocoddyl #605: Feet sliding in Talos simulation

edantec, 2019-11-22:
[Talos_reduced.py.zip](https://github.com/loco-3d/crocoddyl/files/3880278/Talos_reduced.py.zip)

Hello,

I don't know if it's a bug or feature, but I got a problem with sliding feet when trying to simulate a movement in which Talos stands still while reaching a target with its left hand. In this problem the feet should not be moving, but they end up  interpenetrating each other.

I use the DDP method to solve an optimisation problem with various weights (state and control regularization, goal tracking, CoM tracking). Contacts are enabled through 6D contact models. Joints limits are enabled through activation bounds. 

I have provided the simulation file I used. I'm using Python 2.7.

cmastalli, 2019-11-22:
It seems your are having numerical integration errors. This is expected in any DAE, not a bug, and note that the contact dynamics is a DAE.

To mitigate this numerical issue, you should increase the proportional -- Baumgarte  -- gain in your contact models (line 74 and 77 of your code).

For more details about these gains you might want to read the Crocoddyl paper: https://cmastalli.github.io/publications/crocoddyl20unpub.pdf

Yours,
Carlos

cmastalli, 2019-11-22:
Please let me know if this solves your problem.

cmastalli, 2019-11-22:
It seems you are using incorrectly the Baumgarte stabilization (i.e. contact gains in line 74 and 77), so it is expected to see sliding motions. Note that the reference placement is the origin of the inertial frame (i.e. world frame). So you have to options:
 1. Do not use proportional gain (zero value)
 2. Define the properly the reference placement, desired placement of the contact

I suggest to follow 1 because, typically, there is good numerical stability with only the damping term.

For more details about these gains you might want to read the Crocoddyl paper: https://cmastalli.github.io/publications/crocoddyl20unpub.pdf

Yours,
Carlos

cmastalli, 2019-11-22:
I will close this issue since it is expected behavior.

Feel free to reopen it, if it is needed for you

closed 2019-11-22 by cmastalli