loco-3d/crocoddyl #631: Contributing guidelines vs credits

wxmerkt, 2019-12-02:
The `CONTRIBUTING.md` currently lists who has contributed in the past (list of contributors). However, GitHub interprets this special file as the "Contributing guidelines". So when someone opens an issue/pull request they are shown "Contributing guidelines", which aren't guidelines but a list of contributors. Is this intended?

[image]

cmastalli, 2019-12-02:
is this the same https://github.com/loco-3d/crocoddyl/wiki/Developer-Guide?

wxmerkt, 2019-12-02:
Adding the content of the wiki link you said into CONTRIBUTING.md would likely do the trick (or no CONTRIBUTING.md at all). The point is: Having credits in CONTRIBUTING.md is confusing

cmastalli, 2019-12-02:
> Adding the content of the wiki link you said into CONTRIBUTING.md would likely do the trick (or no CONTRIBUTING.md at all). The point is: Having credits in CONTRIBUTING.md is confusing

Roger that! I created an issue #634

proyan, 2020-01-17:
I believe this is done. I'll close this.

closed 2020-01-17 by proyan