loco-3d/crocoddyl #638: selfadjointView may be a bottle neck

jcarpent, 2019-12-03:
https://github.com/loco-3d/crocoddyl/blob/d59b9be9f6c3f90b07dda12e5c14d8889cdc2214/src/multibody/actions/impulse-fwddyn.cpp#L122

From my knowledge of Eigen, `selfadjointView` may lead to slow computations. It may be better to fill the entire mass matrix M and then performing the dense computations.

cmastalli, 2019-12-03:
Thanks for the feedback. I will benchmark it asap!

cmastalli, 2020-04-02:
@proyan you might want to handle this issue.

cmastalli, 2021-09-07:
@jcarpent --- could I ask you (in your experience) when we should use `selfadjointview` (i.e., with which matrix dimension)?

As described here (https://stackoverflow.com/questions/39606224/does-eigen-have-self-transpose-multiply-optimization-like-h-transposeh), `selfadjointview` is not useful for small matrices. But it is not clear (in this post) what matrix dimensions are considered small