loco-3d/crocoddyl #639: Memory allocation may be avoided

jcarpent, 2019-12-03:
https://github.com/loco-3d/crocoddyl/blob/d59b9be9f6c3f90b07dda12e5c14d8889cdc2214/src/multibody/actions/impulse-fwddyn.cpp#L127

This former expression will lead to a temporary evaluation of `f_partial_da * d->impulses->dv0_dq`.

cmastalli, 2019-12-03:
I am using `noalias()`, did you notice it? There is two matrices multiplication, not three!

It shouldn't allocate memory as explained here: https://github.com/libigl/eigen/blob/master/Eigen/src/Core/NoAlias.h#L83

I will close this issue, feel free to reopen it

closed 2019-12-03 by cmastalli

jcarpent, 2019-12-03:
It seems that the Eigen doc is not up to date, and does not track such kind of information contained in the source code of Eigen ;)