loco-3d/crocoddyl #641: Consider passing the ouput as argument

jcarpent, 2019-12-03:
https://github.com/loco-3d/crocoddyl/blob/d59b9be9f6c3f90b07dda12e5c14d8889cdc2214/src/core/optctrl/shooting.cpp#L103

In the former code, at each time the rollout is called, there are some dynamic memory allocations. This can be avoided as all the dimensions are know before hand. You might consider passing the output as argument.

cmastalli, 2019-12-03:
This is an optional method, and not the recommended one. We have another rollout function that doesn't allocate memory: https://github.com/loco-3d/crocoddyl/blob/d59b9be9f6c3f90b07dda12e5c14d8889cdc2214/src/core/optctrl/shooting.cpp#L82

I will close this issue, feel free to reopen it.

closed 2019-12-03 by cmastalli

jcarpent, 2019-12-03:
Sounds fine.