loco-3d/crocoddyl #645: CI fails in Debug mode for Xenial after updating Pinocchio to v2.2.1

cmastalli, 2019-12-05:
The previous merged PR #642 passed the test with v2.1.9 as you can see here:
 - https://travis-ci.org/loco-3d/crocoddyl/jobs/620066192?utm_medium=notification&utm_source=github_status

but it fails during merging to the devel branch:
 - https://travis-ci.org/loco-3d/crocoddyl/jobs/620142296?utm_medium=notification&utm_source=github_status

The only difference between these two jobs is the version of Pinocchio.

I investigated a little more about it, and it seems to be related to the changes in the `pinocchio::Model` and/or `pinocchio::Data`. Concretely, the error appears when running the cost-sum, and this is the message that I encountered:
```
python: /usr/local/include/pinocchio/algorithm/frames.hxx:20: void pinocchio::updateFramePlacements(const pinocchio::ModelTpl<Scalar, Options, JointCollectionTpl>&, pinocchio::DataTpl<Scalar, Options, JointCollectionTpl>&) [with Scalar = double; int Options = 0; JointCollectionTpl = pinocchio::JointCollectionDefaultTpl]: Assertion `model.check(data) && "data is not consistent with model."' failed.
```

@jcarpent I need your help, you might know what it is going on. This only appears on Debug and Xenial. Note that Xenial uses c++98 standard.

@wxmerkt what do you suggest to do? enforced Pinocchio v2.1.9? or survive with it until solution?

jcarpent, 2019-12-05:
To simplify the debugging, can you isolate the model, the reference configuration used?

jcarpent, 2019-12-05:
@cmastalli Can you also give me the write access to the crocodyl. I will force push on your PR for debugging ;)

cmastalli, 2019-12-05:
The PR was merged two days ago. You can test it using the devel branch.

wxmerkt, 2019-12-06:
Is 2.2.1 already on robotpkg?

wxmerkt, 2019-12-06:
I want to make it a priority to switch to C++14 by default on both Xenial and Bionic. Note that when upgrading minimum required versions... if they are not yet fully synced on robotpkg, Travis will fail

jcarpent, 2019-12-06:
@wxmerkt Why do want to switch to C++14?

jcarpent, 2019-12-06:
@cmastalli Everything goes fine for Crocoddyl on my mac, with Pinocchio 2.2.1.

cmastalli, 2019-12-06:
> @cmastalli Everything goes fine for Crocoddyl on my mac, with Pinocchio 2.2.1.

It is the same for Ubuntu 18.04, the problem is Ubuntu 16.04 and in Debug mode.

jcarpent, 2019-12-06:
Maybe the bug is comming from robotpkg. I would suggest to test it with a fresh 16.04 envrionment without using robotpkg.

wxmerkt, 2019-12-06:
> @wxmerkt Why do want to switch to C++14?

C++98 is an unnecessary limitation and there is no reason to stay with it and without the new language features. We are resorting to unnecessary workarounds and need to be super careful not to introduce bugs that would be handled by more modern features.

cmastalli, 2019-12-06:
> Maybe the bug is coming from robotpkg. I would suggest to test it with a fresh 16.04 environment without using robotpkg.

I don't think the problem is robotpkg. I tried in my Ubuntu 16.04 with everything installed by source.

jcarpent, 2019-12-06:
What is the output then?

cmastalli, 2019-12-06:
> > @wxmerkt Why do want to switch to C++14?
> 
> C++98 is an unnecessary limitation and there is no reason to stay with it and without the new language features. We are resorting to unnecessary workarounds and need to be super careful not to introduce bugs that would be handled by more modern features.

We don't know if c++98 is the source of the problem.

cmastalli, 2019-12-06:
> What is the output then?

It's written in the description of this issue

cmastalli, 2019-12-06:
> I want to make it a priority to switch to C++14 by default on both Xenial and Bionic. Note that when upgrading minimum required versions... if they are not yet fully synced on robotpkg, Travis will fail

This is not the right issue to discuss about it. I suggest to write an issue where each person can give a feedback.

In any case, I am not sure if we should go for c++14. Crocoddyl code should be flexible, it's always user choice the compilation mechanism. We have made an effort to support c++98, I don't see yet a value to throw it at this point of the development.

jcarpent, 2019-12-06:
> It's written in the description of this issue

@cmastalli This is not clear at all: can you provide a small snipset to reproduce it in a descent manner?

jcarpent, 2019-12-11:
v2.2.2 is now available. @nim65s Do you have a timeline in mind to push it on robotpkg?

nim65s, 2019-12-11:
I should be able to do that tomorrow :)

cmastalli, 2019-12-11:
Just to clarify -- I am not sure this will fix rhis

nim65s, 2019-12-19:
The issue was on robotpkg side, between C++98 & C++11.

The fix is straightforward, and will be applied on robotpkg. I'll also try to ensure this whole kind of errors will not occur again, robotpkg wide.

I should have been able to look at this on this sooner and prevent you from loosing your time on such stuff, sorry for that.

closed 2019-12-19 by nim65s

cmastalli, 2019-12-19:
When is going to be available? Could you share here the issue reported in robotpkg?

> The issue was on robotpkg side, between C++98 & C++11.
>
> The fix is straightforward, and will be applied on robotpkg. I'll also try to ensure this whole kind of errors will not occur again, robotpkg [...]

When is going to be available? Could you share here the issue reported in robotpkg?

nim65s, 2019-12-19:
It will be available tonight. I'll share here the issue I will report on robotpkg, but I'll need a bit more time to report that and think about the solution, so this might wait tomorrow.

wxmerkt, 2019-12-20:
What was the nature of the C++98 and C++11 conflict?