loco-3d/crocoddyl #648: Which quantity do you want to compute here?

cmastalli, 2019-12-06:
Which quantity do you want to compute here?

_Originally posted by @jcarpent in https://github.com/loco-3d/crocoddyl/pull/644_

wxmerkt, 2019-12-06:
Can we please use more specific issue titles? :-)

proyan, 2019-12-10:
I'll answer here instead of https://github.com/loco-3d/crocoddyl/pull/644#discussion_r354490593.

We are computing the derivatives of the residue (hg-href). The computation of hg in pinocchio is done in the com frame of reference. The computation of the derivatives is then also done in the same frame.

When you want to compute the derivative, you are trying to do lim (dq->0) (hg(q) - hg(q+dq))/dq. Since hg is computed in the CoM frame of reference, there are two changes that need to be accounted for in hg(q+dq). 

1) First is because of the change in hg _because of_ the change in configuration (q+dq). This is done in pinocchio.

2) Second is the change in hg _because of_ the change in CoM frame of reference _because of_ the change in configuration (q+dq). This is what we are doing here.

This was validated in the unittests by finite differencing, which I is here: https://github.com/loco-3d/crocoddyl/blob/master/unittest/test_costs.py#L80

jcarpent, 2019-12-10:
@proyan I think you're making some confusions here:
- `dAg/dq` expressed the variations of the centroidal momentum due to the change in the configuration (taking into account the change of the center of mass positions of all the bodies).
This is true in any with respect to any Plücker coordinate systems.
- the second term you're computing is with respect to the motion of Plücker coordinate systems.

In other words, the Jacobian you're computing cannot be moved thanks to `SE3.act` for instance. And the term you have added can be removed if all the quantities you expressed are done in the same coordinate systems, e.g, the frame aligned with the world and centered on the CoM ;)

jcarpent, 2019-12-10:
@proyan To conclude the previous discussion, you don't have to translate the reference frame by a small amount. It is like spatial vs classic acceleration ;)

proyan, 2019-12-10:
@jcarpent 
I don't think #650 is related here. I'm not sure why that error is happening, but it is not because of this topic.

I know that we are taking care of the moving reference frame when doing derivatives. However, what we actually need to get is the descent direction, and this includes the movement of the reference frame as well. This, as you mentioned, is the difference between the spatial and the classical accelerations as you put it, or th Ring and Dot derivatives as Featherstone puts it.

I've added a simple python script to demonstrate what the confusion is. 

```
import pinocchio
from pinocchio.utils import rand, zero
import numpy as np
EPS = 1e-8

urdf_filename = "<YOUR URDF FILE PATH HERE>.urdf"

# Load the urdf model
rmodel    = pinocchio.buildModelFromUrdf(urdf_filename)

# Create data required by the algorithms
rdata     = rmodel.createData()

q0 = pinocchio.randomConfiguration(rmodel)
v0 = rand(rmodel.nv)
a0 = rand(rmodel.nv)

def df_dq(model, func, q, h=EPS):

    dq = zero(model.nv)
    f0 = func(q)
    res = zero([len(f0), model.nv])
    for iq in range(model.nv):
        dq[iq] = h
        res[:, iq] = (func(pinocchio.integrate(model, q, dq)) - f0) / h
        dq[iq] = 0
    return res

def fn(q):
    pinocchio.computeCentroidalMomentum(rmodel, rdata, q, v0)
    return rdata.hg.vector

#FINITE DIFFERENCING:

dhg_dq = df_dq(rmodel, fn, q0)

#FROM PINOCCHIO:

dhg_dq_pin, _, _, _ = pinocchio.computeCentroidalDynamicsDerivatives(rmodel, rdata,
                                                                     q0, v0, a0)

#After accounting for the moving coordinate system:

pinocchio.computeCentroidalMomentum(rmodel, rdata, q0, v0)
pinocchio.ccrba(rmodel, rdata, q0, v0)
pinocchio.jacobianCenterOfMass(rmodel, rdata, q0)
pinocchio.computeRNEADerivatives(rmodel, rdata, q0, v0, a0)
dhg_dq_act = dhg_dq_pin.copy()

for i, Ji in enumerate(rdata.Jcom.T):
    dhg_dq_act[3:, i] -= np.matrix(np.cross(np.array(Ji).squeeze(),
                                            np.array(rdata.hg.linear).squeeze())).T

print "The ring notation norm", np.linalg.norm(dhg_dq_act-dhg_dq)
print "The dot notation norm", np.linalg.norm(dhg_dq_pin-dhg_dq)

```

Here, what we get from pinocchio is the `dhg_dq_pin`, which is similar to Equation 2.39 of Featherstone. It is the derivative in the CoM frame. What we want to do is to take into account the movement of the CoM frame as well, which means we are not interested by Equation 2.39, but Equation 2.40

cmastalli, 2019-12-10:
> @jcarpent
> I don't think #650 is related here. I'm not sure why that error is happening, but it is not because of this topic.
>
> I know that we are taking care of the moving reference frame when doing [...]

There is no error, this unittest is passing

jcarpent, 2019-12-10:
You have an error here:
```
res[:, iq] = (func(pinocchio.integrate(model, q, dq)) - f0) / h
```
When you do finite differences over Lie groups, you have to be sure that you're expressing all the terms in the same frame. Here, `f0` and `func(pinocchio.integrate(model, q, dq))` are not expressed at the same point, explaining the difference above. 

What it is not clear to me is: do you really need to translate your Jacobians inside the DDP itself? I'm not sure of that.

proyan, 2019-12-10:
I agree, but in DDP what we are eventually trying to find lies in different frames, since the CoM frame also evolves. Definitely it would be easier if all references were in the global frame, but that would introduce additional computations.

I don't really see an alternative to doing this here. Eventually we need to have our own compute-all-terms, and then perhaps we can simplify the translations. Untill then we need to adapt with Pinocchio's outputs.

jcarpent, 2019-12-10:
No, you can expresse all the quantities at the origin. You just need to translate them ;)

proyan, 2019-12-10:
So I would translate hg and href to origin. Then I would translate the dhg to origin. And then I would translate everything back to local frame to communicate with centroidal solver?

jcarpent, 2019-12-10:
Yep. It should be something like that. You can be even more efficient. We can discuss that by Skype if you want.

cmastalli, 2019-12-10:
Is there a reason of why the centroidal solver expressed the `href` in the local frame?

For me, it makes sense to have everything expressed at the origin.

jcarpent, 2019-12-10:
Yes, and not. We are following the original notations of Orin et al. to express the centroidal wrench around the CoM (which is invariant by translation).
But, we may think to have the choice of expressing at another reference frame of course. We can add an optional argument to these functions to express at the origin too.

jcarpent, 2019-12-10:
@cmastalli I think in fact your question was asked to @proyan.
My answer makes no sense then. I will remove it.

proyan, 2019-12-10:
Yes, it makes sense. I think an argument, if you are receptive, of adding a ReferenceFrame.World/Local might be useful here.

I'll do the changes next week.

jcarpent, 2019-12-11:
It will be much more something like WORLD/LOCAL_WORLD_ALIGNED ;)

cmastalli, 2020-03-05:
@proyan I have merged the cost unittest, and effectively the centroidal-momentum is not passing.

Do you have time to solve this issue?

cmastalli, 2020-06-12:
> No, you can express all the quantities at the origin. You just need to translate them ;)

@jcarpent let me understand this statement. We need to transform the coordinate vector that expresses the spatial momentum in frame coordinates to a Cartesian vector, is that correct?

If so, then my understanding is that the force conversion to Cartesian vector is as follows
```python
import pinocchio
...
c = frame_id
c_m = pinocchio.Force.Random() # spatial force defined in the frame coordinate c
oMff = pinocchio.SE3(data.oMf[frame_id].rotation, np.zeros(3))

o_m_wf = oMff.act(c_m)
print("linear momenta", o_m_wf.linear)
print("angular momenta", o_m_wf.angular)
```
right?

But I don't know how to convert the momenta derivatives

jcarpent, 2020-06-12:
Same strategy.

cmastalli, 2020-06-12:
@jcarpent  I have another general question for you.

I don't understand the motivation behind tracking a spatial vector in frame coordinates. Instead I would expect to build a cost function that tracks a Cartesian vector because the users can easily define a reference in the Cartesian space. Is there something that I missed here?

I asked myself the same question for the force tracking cost in ongoing PR #788.

jcarpent, 2020-06-12:
It really depends on your application. From the local Frame, it might be a sensor based tracking.

cmastalli, 2020-06-12:
> It really depends on your application. From the local Frame, it might be a sensor based tracking.

Then, should we define an API cost that accepts all options?, e.g. for the centroidal momenta cost will be like:
  - spatial momenta expresses in local coordinates (body coordinates)
  - spatial momenta expresses in frame coordinates (CoM frame coordinates)
  - Cartesian momenta expresses in the world frame

cmastalli, 2020-06-12:
Or should we write independent cost functions?

closed 2021-08-25 by cmastalli