loco-3d/crocoddyl #655: Boost tests with the same name

nim65s, 2019-12-19:
Hi,

According to [Boost Test Changelog](https://www.boost.org/doc/libs/1_72_0/libs/test/doc/html/boost_test/change_log.html):
>  Adding test cases with the same name to the same test suite is reported as an error. This impacts template and parametrized test cases, as well as manually registered tests. Make sure you have no duplicate names. 

This is a breaking change introduced in Boost v1.67.

In Crocoddyl, this raises issues in 5 test files, eg.:
```
Test setup error: test unit with name 'boost__bind(&test_kkt_dimension, ActionModelTypes__all[action_type], nb_running_models)' registered multiple times in the test suite 'test_solversTest'
```

A quick fix for Boost >= v1.69 (as it uses `BOOST_TEST_CASE_NAME`) is provided in https://github.com/nim65s/crocoddyl/commit/533c3dd5a9be55aa20dcf700198b0b36197c2e52 (tested with boost 1.71)

Another solution might be to use `BOOST_PARAM_TEST_CASE`, which seems to be available since [v1.59](https://www.boost.org/doc/libs/1_59_0/libs/test/doc/html/boost_test/utf_reference/test_org_reference/test_org_boost_test_case_parameter.html), but not in [v1.58](https://www.boost.org/doc/libs/1_58_0/libs/test/doc/html/boost_test/utf_reference/test_org_reference/test_org_boost_test_case_parameter.html).

So we have to define which versions of Boost we want to support. For reference:
- [Debian 8](https://packages.debian.org/fr/jessie/libboost-all-dev): 1.55
- [Ubuntu 16.04](https://packages.ubuntu.com/xenial/libboost-all-dev): 1.58
- [Debian 9](https://packages.debian.org/fr/stretch/libboost-all-dev): 1.62
- [Ubuntu 18.04](https://packages.ubuntu.com/bionic/libboost-all-dev): 1.65
- [Debian 10](https://packages.debian.org/fr/buster/libboost-all-dev) & [Ubuntu 20.04](https://packages.ubuntu.com/focal/libboost-all-dev): 1.67
- [brew (OSX)](https://formulae.brew.sh/formula/boost) : 1.71

So the question is : which version of Crocoddyl should support which version of Boost.

The current version of Crocoddyl doesn't support Boost >= v1.67, and therefore not Debian 10, not Ubuntu 20.04, and not OSX.

And if we want to add support for those plateforms without dropping support for Boost < 1.59 and Ubuntu 16.04, I think we will have to make a macro that will check boost version to declare those tests in a compatible way.

jcarpent, 2019-12-19:
#647 has changed the behavior of the test (for the ones existing). I think following the rules will be sufficient, without the worry of the Boost.Version.

nim65s, 2019-12-19:
Ok, then I'll add #647 in the v1.0.0 release, thanks @jcarpent !

cmastalli, 2019-12-19:
> #647 has changed the behavior of the test (for the ones existing). I think following the rules will be sufficient, without the worry of the Boost.Version.

@nim65s could you test this solution? If it works, it will simplify the code

cmastalli, 2019-12-19:
> Ok, then I'll add #647 in the v1.0.0 release, thanks @jcarpent !

Do we need to add it for v1.0.0? I would prefer to tackle for next release, my point is I have concerns about the Pinocchio data issue in 16.04 Debug mode.

nim65s, 2019-12-19:
As you prefer, but a v1.0.0 which doesn't work on OSX might be a bit strange.

cmastalli, 2019-12-19:
> As you prefer, but a v1.0.0 which doesn't work on OSX might be a bit strange.

If we fix the issue with latest Pinocchio, then I don't have any opposition to include it in v1.0.0. @nim65s could you spend some of your time to fix it?

cmastalli, 2020-03-05:
In #705, I propagated the #647 changes.

This should fix the problem.

closed 2020-03-05 by cmastalli