loco-3d/crocoddyl #666: Rollout with IAMEuler diverges

skleff1994, 2020-01-16:
I implemented a `DAM` for the point mass model (following the [cartpole tutorial](https://gepgitlab.laas.fr/loco-3d/crocoddyl/blob/master/examples/notebooks/cartpole_swing_up.py)) and used `DAMNumDiff` and `IAMEuler` to integrate my model. For sanity check, I perform a forward rollout with initial position 0, initial velocity 1 m/s and control sequence us = 0. But `ShootingProblem.rollout(us)` results in a weird diverging trajectory (blue curve in the plot below) as if the numerical integration scheme was unstable. Under zero input, I was indeed expecting to observe a constant velocity and a linearly increasing position as represented by the green curve. 

[image]

Here's my code : 

```
class DifferentialActionModelPointMass(crocoddyl.DifferentialActionModelAbstract):
    def __init__(self):
        crocoddyl.DifferentialActionModelAbstract.__init__(self, crocoddyl.StateVector(2), 1, 3) # nu = 1, nr = 3
        self.unone = np.zeros(self.nu)
        self.costWeights = [1, 1, 1]

    def calc(self, data, x, u=None):
        if u is None: u=self.unone

        # State and control variables
        p, v = np.asscalar(x[0]), np.asscalar(x[1])
        f = np.asscalar(u[0])

        # Compute next state
        data.xout = np.matrix([v, f]).T

        # Computing cost residual and value
        data.r = np.matrix(self.costWeights * np.array([p, v, f])).T
        data.cost = .5*np.asscalar(sum(np.asarray(data.r)**2))

# Create model
pointmassDAM = DifferentialActionModelPointMass()
pointmassData = pointmassDAM.createData()

# Create derivatives
pointmassND = crocoddyl.DifferentialActionModelNumDiff(pointmassDAM, True)
pointmassDataND = pointmassND.createData()

# Integrate the model 
dt = 1e-2
pointmassIAM = crocoddyl.IntegratedActionModelEuler(pointmassND, dt)

# Create terminal model 
terminalPointmassDAM = DifferentialActionModelPointMass()
terminalPointmassND = crocoddyl.DifferentialActionModelNumDiff(terminalPointmassDAM, True)
terminalPointmassIAM = crocoddyl.IntegratedActionModelEuler(terminalPointmassND, dt)

# Define shooting problem 
x0 = np.matrix([0, 1.]).T
T = 100
problem = crocoddyl.ShootingProblem(x0, [pointmassIAM]*T, terminalPointmassIAM)

# Integrate (rollout)
us = [ pinocchio.utils.zero(pointmassIAM.differential.nu) ]*T
xs = problem.rollout(us)
```
I am using Crocoddyl v1.0.0. Am I doing something wrong here?

cmastalli, 2020-01-17:
You have a problem in the the point-mass DAM. `data.xout` has to be the acceleration of the point-mass system (see here: https://gepgitlab.laas.fr/loco-3d/crocoddyl/blob/master/examples/notebooks/cartpole_swing_up.py#L44).

Note that we have developed a sympletic Euler integrator (not just Euler one). Our integrator preserves the geometric of the system by performing an implicit integration in the configuration space.

bpltdm, 2020-01-17:
Ok I see! Yes that helps, thank you !

cmastalli, 2020-01-17:
I will close this issue.

Feel free to re-open it, if it is needed!

closed 2020-01-17 by cmastalli