loco-3d/crocoddyl #672: Create a non-const reference to get_models in solver-base

proyan, 2020-01-24:
I would like to be able to update the models once they have been created inside the solver class. If you are running the MPC, it is counter-productive to re-define the entire problem again and again just for a small change in one feature.

If you are okay, I can push the changes immediately

wxmerkt, 2020-01-24:
I've asked for a similar API change before but was told that this requires to make sure that the data matches the updated models which would require further added checks. I.e., such a non-const change needs to include data checks in the calc/calcDiff calls.
An alternative would be to update the references in the costs etc. (which is what we currently do).

proyan, 2020-01-24:
>  An alternative would be to update the references in the costs etc. (which is what we currently do).

Could you elaborate please? Once the SolverModel object has been created, it has the CostModel and the cost references inside the action models. In order to update the references without re-creating the problem, you would need to access and modify the action models which are in SolverModel. How do you do that?

wxmerkt, 2020-01-24:
An example from our MPC code is e.g.:
```python
new_cref = ddp.problem.terminalModel.differential.costs.costs['comTrack'].cost.cref.copy()
new_cref[0] = 0.1
ddp.problem.terminalModel.differential.costs.costs['comTrack'].cost.cref = new_cref
```

proyan, 2020-01-24:
I see. I'm working in c++, and I think the only way around for me would be to save a pointer to the cost reference which is being shared, and update it this way.

I think I can make do, but I don't think this would be the right approach. This would still avoid the data checks that supposedly are preventing a proper solution. Moreover, this is limiting to specific problem types. For example if we have the contact planner in the loop, we might want to change the differential action model to add or remove contacts. Or perhaps change the DAM Model types.
I don't like this being used as a proper solution.

I am more in favor of having a data-checking function before we call solver.solve in debug mode. Otherwise If a user wants to update the model, he should be allowed to do that.

proyan, 2020-01-24:
Thanks for your solution @wxmerkt

cmastalli, 2020-01-25:
This is not the right way of doing it. If we need to change the model, then we have to interact with the shooting problem object directly.

However, I agree that we should have the following changes:
 - remove `get_models` and `get_datas` since it contains in the shooting problem
 - give freedom to set a different shooting problem

I will provide a solution soon!

**Additional comment**
I am currently working on a MPC formulation that will works for quadrupeds, bipeds, centaurs and humanoids. This implementation will handle all these details.

cmastalli, 2020-01-25:
I will close this issue, feel free to open it.

proyan, 2020-01-26:
>This is not the right way of doing it. If we need to change the model, then we have to interact with the shooting problem object directly.

The issue is not whether we change the model through solver class or the shooting problem class. I think the user should simply _have_ the ability to change the model. Currently, every access to any element of the model is const. All elements of the ddp class are const. I find it really limiting.

cmastalli, 2020-01-27:
I will provide a solution soon (I will let you know), through shooting problem interface.

Note that this proposition is not correct since we have decoupled problem and solvers.

wxmerkt, 2020-03-10:
I'd like to revisit this topic for the C++ API on order to be able to update cost weights. What is the current status of discussions to enable this?

1. Add a setter to every const-item?
2. Add a non-const reference getter?

Example that fails:
```cpp
my_differential_model->get_costs()->get_costs().find("comTrack")->second.weight = 1.;
```

cmastalli, 2020-03-10:
I am in favor of with adding  a setter function (option 1).

However, it shouldn't fail that example code if you `my_differential_model` is type of `DifferentialActionModelContactFwdDynamics` or `DifferentialActionModelFreeFwdDynamics`.

Note that `get_costs` appears only in multibody DAM, not in the base class (i.e. `DifferentialActionModelAbstract`).

I am in favor (and almost to implement) to add the costs (and in future constraints base classes) inside the DAM abstraction. The main reason is that if we include constraints in our solvers, then they should be able to retrieve this information.

For adding the cost in the DAM abstraction, I need to remove the Pinocchio dependency which is a fairly simple task (in my TODO list).

wxmerkt, 2020-03-10:
I tried Option 1. It fails because the returned container map (`get_costs()`) is a const-ref. As thus, we cannot change the weight on an element belonging to that container - unless it's a shared_ptr.

cmastalli, 2020-03-10:
Then there are two options:
 1. Define CostItem as shared ptr, i.e.
```cpp
typedef std::map<std::string, boost::shared_ptr<CostItem> > CostModelContainer;
```
 2. Return a nonconst-ref of the `CostModelContainer`, i.e.
```cpp
CostModelContainer& get_costs() const;
```

For me, it makes more sense option 1.

wxmerkt, 2020-03-10:
I am not a fan of changing the API too often - and we do have some inconsistencies with raw-references vs pointers already (e.g. with Pinocchio models). Happy with either solution.

@proyan, what are your thoughts?

proyan, 2020-03-10:
> ```c++
> my_differential_model->get_costs()->get_costs().find("comTrack")->second.weight = 1.;
> ```

This sounds more intuitive to me than setters. Moreover, creating setters means that you need to create a new object and set it as the old one, and I don't like the idea of doing that. I prefer tinkering with the objects that are already defined.

cmastalli, 2020-03-10:
The user solution is quite simple. You store each cost model in your MPC class, you really don't need to interact with them through the action models as they are shared pointers.

wxmerkt, 2020-03-10:
> This sounds more intuitive to me than setters. Moreover, creating setters means that you need to create a new object and set it as the old one, and I don't like the idea of doing that. I prefer tinkering with the objects that are already defined.

This is a valid point, as long as we do not mess up the original object. For costs, that should be fine. I wasn't happy with the API in Python where we needed to create a full copy to modify a single term to then assign it back via the setter.

As long as we modify properties within the shared pointer, I am also happy with that solution.

I'd like to wrap up this discussion shortly to go ahead with modifying the code to support this feature. What other arguments should we consider?

wxmerkt, 2020-03-10:
Proposal: 
(1) We change the `get_costs()` return type to non-const-ref now.
(2) If we want to revisit to a shared_ptr interface, we do that at a later stage, with proper transition plan and announcement, and major version increase. The `Cost` itself is already a shared_ptr, just not the `CostItem`.

This is the minimal and least disruptive effort to get this resolved. What are your thoughts?

cmastalli, 2020-03-10:
We don't need to pursue any of the options since the users can store the cost models and modify by your own.

I don't like option (1), the getter function are just for retrieving things. If we pursue option 1, then there is no consistency in the code.

wxmerkt, 2020-03-10:
> We don't need to pursue any of the options since the users can store the cost models and modify by his/her own.

Except the weight!

cmastalli, 2020-03-10:
I believe the solution is `shared_ptr` option, this is how it is done along the entire library. It would create tiny changes in the internal API, but the other option creates inconsistencies and we need to propagate in the whole library.

Since nobody can't set cost properties as you're pointing out, then nobody will be affected by the used of `shared_ptr`.

We still expect to have changes in the API if they're justify. I think this is the case.

Let's pursue this direction :)

wxmerkt, 2020-03-10:
Are you tackling it in an atomic PR, and if so, what is your timeline? Otherwise I can pick it up.

cmastalli, 2020-03-10:
> Are you tackling it in an atomic PR, and if so, what is your timeline? Otherwise I can pick it up.

I could do it this Thursday. Feel free to create an atomic PR if this is not quick enough for you.

cmastalli, 2020-03-20:
I think we could close this issue since it was solved in #712.

If you want to handle more specific changes regarding a similar topic, then let's do it through another issue.

closed 2020-03-20 by cmastalli