loco-3d/crocoddyl #674: We don't need x0 inside the ShootingProblem class

proyan, 2020-01-26:
Hi guys, 

- Currently we have two ways to set the initial state (the first state only) for the ddp problem. We can set it through init_xs[0] inside the Solver class, or we can set it through x0 inside the ShootingProblem class

- Currently x0 inside the shooting problem class is being used for: 
   - to start the ShootingProblem::rollout method.
   - to start the SolverDDP::calc method.

- Currently, ShootingProblem::rollout is not being used at all.

It is quite misleading to have to different ways to set the initial condition. If x0 and init_xs[0] don't have the same initial condition, it gives a giant bug. Moreover, I don't see the use of saving an initial condition inside the ShootingProblem class. Initial conditions should arise only when we are dealing with the ddp.solve() method, not before.

So I recommend removing x0 from the ShootingProblem class, and change the API for `ShootingProblem::rollout`  to `ShootingProblem::rollout(x0, u_init(trajectory) )`

cmastalli, 2020-01-27:
You are misunderstanding things, 'init_xs[0]' is the warm-start point, not 'x0'.

If there is a different between 'init_xs[0]' and 'x0', then it is accounted in the gaps which through the next iterations (i.e. xs_try[i]) is closed; there is not any "giant" bug, it is the expected behavior. Additionally, note that the solvers always use ShootingProblem::'x0'.

I agree it would be nice to use ShootingProblem::rollout inside the solvers, we have already discussed it. This is indeed to be done, but it has to be though well in order to consider infeasible rollouts and adaptive integrators.

proyan, 2020-01-27:
It is not possible to change the initial condition in DDP. DDP updates the control variables, and there is no control that can act on the first state and change the first state. Could you please explain where we are changing the initial state in DDP?

cmastalli, 2020-01-27:
I see clearly your point.

The xs_try[0] is defined given problem_->get_x0() while we allocate data, see:
 - https://github.com/loco-3d/crocoddyl/blob/devel/src/core/solvers/ddp.cpp#L341
which avoids the problem because after accepting the first step we introduce that value.

However, it wouldn't happen in the next calls of 'solve' which occurs in a MPC implementation. Therefore, it is important that user defines the same 'x0' and 'init_xs[0]'. This is not error prone, and it needs to be improved. To solve it, I suggest to rewrite the value of 'xs_try[0]' from 'x0' before starting to iterate.

Is this sounds good for you? If so, could you handle it?

proyan, 2020-01-27:
Unfortunately, to use crocoddyl right now, you would need to 
1) Define x0 in shooting problem
2) Define x_init{0} in ddp class.

And you need to do this in exactly the above order, because if we change the above order, then the problem_->get_x0() call inside allocateData is meaningless.

From my part, I already wasted multiple hours because of this small step, and I'm sorry I don't see the point of saving x0 inside the shooting problem. If the idea is to save the initial condition for shooting, then {x0, u_traj} both form the initial condition. And these are already being taken inside the ddp class. I don't think we should be saving x0 separately inside the shootingproblem and u_traj inside ddp class.

cmastalli, 2020-01-28:
I understand your concerns, and I tend to agree with you. However, I believe there is a clear difference between x0 and x_init[0], and this is especially important for feasibility-prone solvers.

Maybe we should ask ourselves, do we want to warm-start an infeasible initial condition? does it make sense? -- @nmansard, what is your answer?

@proyan if there is no a clear answer for this question. I will propose to follow my suggestion, which it doesn't change API neither. We could still spend time to answer it, to make a final decision.

I could handle it, but after January 31. I am working in a RSS paper.

proyan, 2020-01-28:
> this is especially important for feasibility-prone solvers.
Again, as I said before, you cannot change the initial value of a ddp (or fddp) simulation :)

Just to illustrate, this is what happens to one joint angle, when you run an MPC with different x_init and x0:

[image]

The solver always starts from the x0, even though the value being given in x_init is the correct one. The solver starts from the value which was initially set by x0, and not reset again. Even though the value being given in x_init is being changed. The solver cannot change the initial state. Neither can the feasibility prone ddp.

This is what happens when both x_init and x0 correspond to the same value:

[image]

cmastalli, 2020-01-28:
Both propositions (mine and yours) could fix it, the former will still keep `x0` in shooting problem. The main difference from user perspective is that he/she will require to `set_x0` in the shooting problem.

Thanks for sharing the pic! Looking forward to see the MPC working on Talos!

wxmerkt, 2020-01-28:
> The main difference from user perspective is that he/she will require to set_x0 in the shooting problem.

I think I already exposed `set_x0` for the shooting problem - it's what I use in MPC.

guru-florida, 2020-03-09:
I am wondering if I am encountering this issue. I have adapted the bipedal_walk.py example to use my own robot (my own urdf/srdf). I get successful feasability on the Walking phase 0, but infeasable phase 2, the 'stop' value quickly jumps to over 1e20 in 6 iterations. Gepetto has a good looking first step, it takes a half step with the right, then a full step with the left. At this point the left leg is advanced of the right. Then the next phase wont solve.

This is the example, so x0 is getting set with ddp[i].xs[-1]. I also tried changing xs init line just before solve():
`xs = [humanoid.model.defaultState] * len(ddp[i].models())`
   to this trying to get a closer warm-start:
`xs = [x0] * len(ddp[i].models())`

I am using robotpkg 1.1.0 version...does this include the _xs_try[0] fix?

```
*** SOLVE walking.0 ***
[snip]
iter 	 cost 	      stop 	    grad 	  xreg 	      ureg 	 step 	 feas
  90  4.65299e+03  7.32573e-04  9.88472e-13  1.00000e+03  1.00000e+03   1.0000     1
  91  4.65299e+03  7.37762e-04  1.07712e-11  1.00000e+02  1.00000e+02   1.0000     1
  92  4.65299e+03  7.78778e-04  8.64984e-11  1.00000e+02  1.00000e+02   0.0156     1
  93  4.65299e+03  7.81403e-04  1.00789e-10  1.00000e+03  1.00000e+03   0.0020     1
  94  4.65299e+03  7.04553e-04  1.29299e-11  1.00000e+04  1.00000e+04   0.0020     1
  95  4.65299e+03  6.89624e-04  1.35130e-12  1.00000e+05  1.00000e+05   0.0078     1
  96  4.65299e+03  6.89375e-04  1.31324e-13  1.00000e+04  1.00000e+04   1.0000     1
  97  4.65299e+03  7.03346e-04  1.13222e-12  1.00000e+03  1.00000e+03   1.0000     1
  98  4.65299e+03  7.01491e-04  8.99531e-12  1.00000e+03  1.00000e+03   0.5000     1
  99  4.65299e+03  6.78278e-04  9.61028e-12  1.00000e+03  1.00000e+03   0.0625     1
*** SOLVE walking.1 ***
iter 	 cost 	      stop 	    grad 	  xreg 	      ureg 	 step 	 feas
   0  1.35545e+07  3.72376e+18  6.15932e+07  1.00000e+00  1.00000e+00   0.0020     0
   1  1.35545e+07  3.72379e+18  6.17063e+07  1.00000e+01  1.00000e+01   0.0020     0
   2  1.35545e+07  3.72407e+18  6.28180e+07  1.00000e+02  1.00000e+02   0.0020     0
   3  1.35545e+07  3.72689e+18  7.32870e+07  1.00000e+03  1.00000e+03   0.0020     0
   4  1.35545e+07  3.76370e+18  1.69618e+08  1.00000e+04  1.00000e+04   0.0020     0
   5  1.35545e+07  4.96959e+18  1.04084e+09  1.00000e+05  1.00000e+05   0.0020     0
   6  1.35545e+07  8.77185e+19  8.62345e+09  1.00000e+06  1.00000e+06   0.0020     0
   7  1.35545e+07  7.15304e+21  7.73960e+10  1.00000e+07  1.00000e+07   0.0020     0
   8  1.35545e+07  6.79129e+23  7.44371e+11  1.00000e+08  1.00000e+08   0.0020     0
( solve stops here )
```

cmastalli, 2020-03-09:
@guru-florida I don't think your problem is related with this thread.

I have implemented a fairly simple formulation for the examples. They need to be hand-tuned for each robot (dynamics).

I am currently working in a formulation that will automatize this process.
However, it is not bad idea if you spend time trying to formulate such kind of problems.
You will learn a lot through this process.

guru-florida, 2020-03-10:
@cmastalli Yes, agreed. I've been digging around the biped.py support code, robot loaders, and even in the C++ headers to get API clarity. It's getting clearer but I still have lots to learn. I'll keep plugging away at it.

cmastalli, 2020-03-20:
I will close this issue since it was solved in #677 (the method that I proposed).

@proyan I hope you feel OK with this solution.
I am afraid that it is not possible to go through your method.
Note that we might want to wrap the shooting problem class into off-the-shelf solvers with initial constraint.

closed 2020-03-20 by cmastalli