loco-3d/crocoddyl #678: Installing crocoddyl for python 3.6

julesser, 2020-02-04:
Hi,

I've installed crocoddyl through robotpkg following the instructions within the repo "https://github.com/loco-3d/crocoddyl".

Starting examples e.g. "arm_manipulation.py" works fine. However, setting attributes for plotting or displaying the solution throws an error.

**E.g. for "python3 arm_manipulation.py plot":**

 File "arm_manipulation.py", line 88, in <module>
    crocoddyl.plotConvergence(log.costs, log.u_regs, log.x_regs, log.grads, log.stops, log.steps, figIndex=2)
AttributeError: 'CallbackLogger' object has no attribute 'u_regs'

**And for "python3 double_pendulum.py display":**

  File "double_pendulum.py", line 50, in <module>
    display = crocoddyl.GepettoDisplay(robot, 4, 4, cameraTF, False)
  File "/opt/openrobots/lib/python3.6/site-packages/crocoddyl/__init__.py", line 43, in __init__
    self.addRobot()
  File "/opt/openrobots/lib/python3.6/site-packages/crocoddyl/__init__.py", line 100, in addRobot
    self.robot.loadViewerModel(rootNodeName="robot")
  File "/opt/openrobots/lib/python3.6/site-packages/pinocchio/robot_wrapper.py", line 241, in loadViewerModel
    self.viz.loadViewerModel(*args, **kwargs)
  File "/opt/openrobots/lib/python3.6/site-packages/pinocchio/visualize/gepetto_visualizer.py", line 69, in loadViewerModel
    gui = self.viewer.gui
AttributeError: 'GepettoVisualizer' object has no attribute 'viewer'

Are these errors related to wrong env variables or any idea what could cause this?

Best,
Julian

proyan, 2020-02-04:
Hi Julian,

'GepettoVisualizer' object has no attribute 'viewer' is typically a sign 
that the gepetto-gui/gepetto-viewer-server is not running.
Could you please try to run gepetto-gui in a separate terminal, and then 
run the double_pendulum example?
Rohan

cmastalli, 2020-02-04:
To run the gepetto-viewer, you need to run on another terminal as follows:
```bash
gepetto-gui
```

cmastalli, 2020-02-04:
I will close this issue.
Feel free to add more questions if you still have troubles

closed 2020-02-04 by cmastalli