loco-3d/crocoddyl #680: AttributeError: 'DifferentialActionDataContactFwdDynamics' object has no attribute 'contacts'

julesser, 2020-02-05:
Hi,
`python bipedal_walk.py` successfully runs. 
When adding the `display` argument for visualization, the following error occurs: 

>   File "bipedal_walk.py", line 72, in <module>
    ddp[i].solve(xs, us, 100, False, 0.1)
  File "/opt/openrobots/lib/python3.6/site-packages/crocoddyl/__init__.py", line 158, in __call__
    self.visualization.displayFromSolver(solver)
  File "/opt/openrobots/lib/python3.6/site-packages/crocoddyl/__init__.py", line 91, in displayFromSolver
    fs = self.getForceTrajectoryFromSolver(solver)
  File "/opt/openrobots/lib/python3.6/site-packages/crocoddyl/__init__.py", line 121, in getForceTrajectoryFromSolver
    for key, contact in data.differential.contacts.contacts.items():
AttributeError: 'DifferentialActionDataContactFwdDynamics' object has no attribute 'contacts'

cmastalli, 2020-02-05:
Hi @julesser,

Thanks for showing interests in our software. Hopefully, you could contribute to our effort.

To display any example, you need to launch the gepetto-gui in another terminal.
For doing so, you can to:
 1. install gepetto-viewer with osg-dae plugin (needed for robots with DAE meshes)
 2. type the commant ```gepetto-gui```

I will close this issue. Feel free to open or ask for more questions here!

Quick suggestion -- There are additional examples in the devel branch, soon we will make a new release to robotpkg.

closed 2020-02-05 by cmastalli

julesser, 2020-02-05:
Thanks for the quick following up. 
As you suggested in #678 I already installed the gepetto viewer and successfully could display some examples, e.g. `python3 double_pendulum.py display` works fine. 
Nevertheless  some examples fail with the `display` option, for example the bipedal_walk as mentioned above. 
Any idea on how to properly handle the "contact" attribute error?

guru-florida, 2020-02-09:
I am also getting the same error on biped_walk. ty

cmastalli, 2020-02-09:
Are you running the gepetto viewer? The error message is due to it does not find it.

Additionally, a recent modification in example-robot-data triggers another error. Please update this package to v3.1.2..

New releases are going to appear soon.

julesser, 2020-02-10:
I am running the gepetto viewer and have updated all packages to the newest version, i.e. the robot data to v3.1.2.  
The 'contacts' error message still remains the same.

cmastalli, 2020-02-10:
How did you install gepetto viewer? from robotpkg binaries?
If so, do you have ubuntu 16.04?
Are you having the same display problem in other examples?

I am asking all these questions because I believe gepetto-viewer from Ubuntu 16.04 binaries are not compatible (old version) to the display.

guru-florida, 2020-02-11:
Hi cmastalli. Thanks for your time! I cannot seem to get the biped_walk to work. I have gepettogui running, when running biped_walk I for a second see the talos hips, then it flickers to only its foot(?), then crashes.

- I am using python3.7 on Ubuntu19 on amd64
- I installed via apt install robotpkg-py37-crocoddyl, robotpkg-py37-qt4-gepetto-viewer-corba, robotpkg-osg-dae
- strangely, robotpkg-osg-dae is installed but gepettogui only lists OmniORB Server in plugins...I expected DAE plugin here. FindPlugins button doesnt find it and in fact "dpkg -L robotpkg-osg-dae" doesnt list any .so or python file...only a bunch of doc files and a few include ones. Only a pkgconfig file is installed in the /opt/openrobots/lib dir.
- I did a pprint(vars(data.differential)) on the line 123 error in the output and it returns an empty {} .... there is no contacts.contacts.items() on the object.
- The robotpkg-py37-crocoddyl package does not install the /examples files...so I am running the examples from a github checkout...tried both master and devel branches.

The entire output of bipedal_walk is:
```
guru@Kuba:~/src/crocoddyl/examples$ python3 bipedal_walk.py display plot
*** SOLVE walking ***
iter 	 cost 	      stop 	    grad 	  xreg 	      ureg 	 step 	 feas
   0  1.09774e+06  2.54565e+11  8.12082e+07  1.00000e-02  1.00000e-02   1.0000     1
   1  2.15012e+05  2.35119e+13  2.18849e+06  1.00000e-03  1.00000e-03   1.0000     1
   2  4.72918e+03  6.26126e+12  4.24250e+05  1.00000e-04  1.00000e-04   1.0000     1
   3  2.87295e+03  6.44189e+10  3.72378e+03  1.00000e-05  1.00000e-05   1.0000     1
{}
Traceback (most recent call last):
  File "bipedal_walk.py", line 73, in <module>
    ddp[i].solve(xs, us, 1000, False, 0.1)
  File "/opt/openrobots/lib/python3.7/site-packages/crocoddyl/__init__.py", line 160, in __call__
    self.visualization.displayFromSolver(solver)
  File "/opt/openrobots/lib/python3.7/site-packages/crocoddyl/__init__.py", line 92, in displayFromSolver
    fs = self.getForceTrajectoryFromSolver(solver)
  File "/opt/openrobots/lib/python3.7/site-packages/crocoddyl/__init__.py", line 123, in getForceTrajectoryFromSolver
    for key, contact in data.differential.contacts.contacts.items():
AttributeError: 'DifferentialActionDataContactFwdDynamics' object has no attribute 'contacts'

```

This is the output from the gepetto viewer. The code in bipedal_walk.py suggests that it loads the robot example file, is this correct? Or do I need to first load the talos robot in gepetto? Not sure why the iostream error, file not found maybe?
```
guru@Kuba:~/src/crocoddyl$ gepetto-gui 
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
List of GroupNodes :
   crocoddyl
   gepetto-gui
   hpp-gui
BOX does not exist.: iostream error
BOX does not exist.: iostream error
CYLINDER does not exist.: iostream error
CYLINDER does not exist.: iostream error
CYLINDER does not exist.: iostream error
CYLINDER does not exist.: iostream error
BOX does not exist.: iostream error
Node "world/robot/collisions/leg_left_6_link_0" does not exist.
Node "world/robot/collisions/leg_right_6_link_0" does not exist.
Node "world/robot/collisions/wrist_left_ft_link_0" does not exist.
Node "world/robot/collisions/wrist_left_ft_tool_link_0" does not exist.
Node "world/robot/collisions/wrist_right_ft_link_0" does not exist.
Node "world/robot/collisions/wrist_right_ft_tool_link_0" does not exist.
Node "world/robot/collisions/rgbd_link_0" does not exist.
You must initialize a Geode on world/robot/visuals/base_link_0 to use Alpha
You must initialize a Geode on world/robot/visuals/leg_left_1_link_0 to use Alpha
You must initialize a Geode on world/robot/visuals/leg_left_2_link_0 to use Alpha
You must initialize a Geode on world/robot/visuals/leg_left_3_link_0 to use Alpha
You must initialize a Geode on world/robot/visuals/leg_left_4_link_0 to use Alpha
You must initialize a Geode on world/robot/visuals/leg_left_5_link_0 to use Alpha
You must initialize a Geode on world/robot/visuals/leg_left_6_link_0 to use Alpha
You must initialize a Geode on world/robot/visuals/leg_right_1_link_0 to use Alpha
You must initialize a Geode on world/robot/visuals/leg_right_2_link_0 to use Alpha
You must initialize a Geode on world/robot/visuals/leg_right_3_link_0 to use Alpha
You must initialize a Geode on world/robot/visuals/leg_right_4_link_0 to use Alpha
You must initialize a Geode on world/robot/visuals/leg_right_5_link_0 to use Alpha
You must initialize a Geode on world/robot/visuals/leg_right_6_link_0 to use Alpha

```

guru-florida, 2020-02-11:
FYI  I compiled osg-dae from source and it does in fact create a so file. So it seems the robotpkg-osg-dae package is missing the target artifact. I was able to install the so file but gepetto gui doesnt like it. It finds it but says "unknown error" trying to load it.

osg-dae build steps (in git cloned osg-dae dir):
```
git submodule update --init
mkdir build && cd build
export PKG_CONFIG_PATH=/opt/openrobots/lib/pkgconfig
cmake -DCMAKE_INSTALL_PREFIX=/opt/openrobots  ..
make
sudo make install
```

julesser, 2020-02-11:
- I am using python3.6.9 on Ubuntu18.04
- I installed via apt install robotpkg-py36-crocoddyl, robotpkg-py36-qt4-gepetto-viewer-corba

- The following examples are displayed successfully: 
```
python3 arm_manipulation.py display
python3 double_pendulum.py display
python3 quadrotor.py display
python3 quadrotor_ubound.py display
```

- These one fail with the error message stated above: 
```
python3 bipedal_walk.py display
python3 bipedal_walk_ubound.py display
python3 humanoid_manipulation.py display
python3 quadrupedal_gaits.py display
python3 quadrupedal_walk_ubound.py display
```

julesser, 2020-02-11:
These show even other error messages without the `display` option: 
`python3 boxfddp_vs_boxddp.py`

> Traceback (most recent call last):
>   File "boxfddp_vs_boxddp.py", line 46, in <module>
>     boxfddp = crocoddyl.SolverBoxFDDP(
> AttributeError: module 'crocoddyl' has no attribute 'SolverBoxFDDP'

```
- python3 humanoid_manipulation_ubound.py
- python3 humanoid_taichi.py
```

>   File "humanoid_manipulation_ubound.py", line 126, in <module>
>     ddp = crocoddyl.SolverBoxFDDP(problem)
> AttributeError: module 'crocoddyl' has no attribute 'SolverBoxFDDP'

cmastalli, 2020-02-11:
I think you guys might have an installation issue.
Probably it is related to Python version.

Please install again Crocoddyl and double check that you're using the right version of Python.
For instance, it could be that you install Crocoddyl using Python 2.

julesser, 2020-02-11:
I've tried some modifications regarding the python version to use: 

- Reinstalled binaries for **python3.6.9** 
(sudo apt install robotpkg-py36-crocoddyl, robotpkg-py36-qt4-gepetto-viewer-corba) and call example via `python3 bipedal_walk.py display`. 
- Completely removed robotpkg and installed binaries for **python2.7.17**
(sudo apt install robotpkg-py27-crocoddyl, robotpkg-py27-qt4-gepetto-viewer-corba) and call example via `python bipedal_walk.py display`.
Error message concerning  "no attribute contacts'' still appears.

cmastalli, 2020-02-11:
The problem we didn't release this code yet. So you need:
 1. to remove `robotpkg-py36-crocoddyl`
 2. to install crocoddyl by source

In any case I am trying to make a new release of Crocoddyl. It could take around a week from now.

julesser, 2020-02-12:
Ok, thanks for the effort put in. In order to get things running now, I
1. Removed `robotpkg-py36-crocoddyl`
2. Will install crocoddyl by source executing the following steps in the cloned crocoddyl dir
```
git submodule update --init
mkdir build && cd build
export PKG_CONFIG_PATH=/opt/openrobots/lib/pkgconfig
cmake -DCMAKE_INSTALL_PREFIX=/opt/openrobots  ..
make
sudo make install
```
The related question is: 
How can I make shure to install it with the python 3.6 bindings?

julesser, 2020-02-12:
Following the build process stated above for the latest `devel` branch of crocoddyl, the `make` command returns an **error**:

```
.....
-DOCTOMAP_MINOR_VERSION=9 -DOCTOMAP_PATCH_VERSION=0 -o CMakeFiles/crocoddyl.dir/multibody/actions/impulse-fwddyn.cpp.o -c /home//########/crocoddyl/src/multibody/actions/impulse-fwddyn.cpp
//########/crocoddyl/src/multibody/actions/impulse-fwddyn.cpp: In constructor ‘Eigen::internal::evaluator_base<Eigen::CwiseBinaryOp<Eigen::internal::scalar_product_op<double, double>, const Eigen::Transpose<const Eigen::Block<const Eigen::Transpose<Eigen::Block<Eigen::Matrix<double, 6, -1, 0, 6, -1>, 6, 1, true> >, 1, 6, true> >, const Eigen::Block<const Eigen::Matrix<double, 6, 6, 0, 6, 6>, 6, 1, true> > >::evaluator_base()’:
/home//########/crocoddyl/src/multibody/actions/impulse-fwddyn.cpp:184:1: internal compiler error: Segmentation fault
 }  // namespace crocoddyl
 ^
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
src/CMakeFiles/crocoddyl.dir/build.make:1409: recipe for target 'src/CMakeFiles/crocoddyl.dir/multibody/actions/impulse-fwddyn.cpp.o' failed
make[2]: *** [src/CMakeFiles/crocoddyl.dir/multibody/actions/impulse-fwddyn.cpp.o] Error 1
make[2]: Leaving directory '/home//########/crocoddyl/build'
CMakeFiles/Makefile2:2587: recipe for target 'src/CMakeFiles/crocoddyl.dir/all' failed
make[1]: *** [src/CMakeFiles/crocoddyl.dir/all] Error 2
make[1]: Leaving directory '/home//########/crocoddyl/build'
Makefile:143: recipe for target 'all' failed
make: *** [all] Error 2

```

cmastalli, 2020-02-12:
> How can I make shure to install it with the python 3.6 bindings instead of python 2.7?

You need to properly define the `PYTHONPATH`. The you need to run the right python interpreter each time that you call a script.

cmastalli, 2020-02-12:
> .....
> -DOCTOMAP_MINOR_VERSION=9 -DOCTOMAP_PATCH_VERSION=0 -o CMakeFiles/crocoddyl.dir/multibody/actions/impulse-fwddyn.cpp.o -c /home//########/crocoddyl/src/multibody/actions/impulse-fwddyn.cpp [...]

We are unit testing our code under your setup: Ubuntu 18.04. Do you install multiple versions of Eigen? Which version did you install?
@nim65s do you any idea what could be the possible sources of this compilation problem?

julesser, 2020-02-12:
> You need to properly define the PYTHONPATH. The you need to run the right python interpreter each time that you call a script.
Okay, if that's all there is to do I'm on the right track.

julesser, 2020-02-12:
From the previous binary installation of `robotpkg-py36-crocoddyl` the following dependencies are still installed: 
[image]

Additionally I can find  `libeigen3-dev` installed in version 3.3.4-4 (for the case that's relevant for you).

cmastalli, 2020-02-12:
Could be that you are running out of RAM? Did you try to compile few times with a single thread (or less threads)?

cmastalli, 2020-02-14:
@julesser did you solve your compilation issue?
If not, we have just released the binaries of v1.1.0 and contains all the modifications needed for running all examples and notebooks.

guru-florida, 2020-02-14:
Sweet!!! :)  I can confirm the updated release packages (robotpkg/python37) run the Talos biped walking simulation! 

FYI  I just read your whitepaper from last month on using HPP Loco3D + Crocoddyl to create a database memory of joint trajectories for warm starts....very interesting. I had a similar idea I called "shape memory" or kinesthetic memory that is much the same idea. I wasn't sure how I was going to tie the memory into the humanoid dynamics though....so good timing in my case! :)

I do get an error at the end, but completes about 7 leg cycles before doing so, so I am not worried about it.
```
Traceback (most recent call last):
  File "bipedal_walk.py", line 92, in <module>
    plotSolution(talos_legs.model, xs, us, figIndex=1, show=False)
TypeError: plotSolution() got multiple values for argument 'figIndex'

```

julesser, 2020-02-17:
@cmastalli 
Compiling the the most current sources (this morning) worked fine.
Displaying with bipedal walking of the Talos legs runs successful with the updated release packages (robotpkg/python36).
 **Thanks a lot for taking care and your great support :-)**

cmastalli, 2020-02-17:
> Sweet!!! :) I can confirm the updated release packages (robotpkg/python37) run the Talos biped walking simulation!
> 
> FYI I just read your whitepaper from last month on using HPP Loco3D + Crocoddyl to create a database memory of joint trajectories for warm starts....very interesting. I had a similar idea I called "shape memory" or kinesthetic memory that is much the same idea. I wasn't sure how I was going to tie the memory into the humanoid dynamics though....so good timing in my case! :)
> 
> ~I do get an error at the end, but completes about 7 leg cycles before doing so, so I am not worried about it.~ Disregard, call to plotSolution() changed and I didnt update my examples working directory to latest. doh

Crocoddyl is a core toolbox for our EU project called MEMMO (stands for Memory of Motion). For more details visit https://www.memmo-project.eu/.

cmastalli, 2020-02-17:
Happy to know that the code works for both :)