loco-3d/crocoddyl #682: cost function to minimize the acceleration of torso while impact

ddliugit, 2020-02-05:
I am new to crocoddyl. Is there a cost function to minimize the acceleration of torso while robot lands on the ground?

I checked the jumping example. It seems no soft landing after jumping.

cmastalli, 2020-02-07:
Acceleration cannot be penalize, instead we could penalize velocities (or impact velocities).

During landing phases, we often use a impulse dynamics; however, it is also possible to use contact dynamics + high velocity penalization (soft-constraint). For reference about the impulse dynamics see Eq (7): https://cmastalli.github.io/publications/crocoddyl20unpub.pdf.

## Contact dynamic + high velocity penalization
For the latter case, you could use a state cost with a weighted quadratic activation in which 
your highly penalize the torso velocity (and the joint velocities).

## Impulse dynamics
At the moment (devel branch), there is a cost function that penalizes the impact velocity at the CoM (not torso).
In future, I will incorporate the penalization of the whole-body impact velocity.

ddliugit, 2020-02-07:
What's the API name that penalizes the impact velocity at the CoM?

BTW, can you recommend some references if I want to write explicit math representations for model and cost compiled in crocoddyl?

cmastalli, 2020-02-07:
> What's the API name that penalizes the impact velocity at the CoM in impulse dynamics?

`crocoddyl::CostModelImpulseCoM`

> BTW, can you recommend some references if I want to write explicit math representations for model and cost compiled in crocoddyl?

You could derive action models (or differential action models) and costs using Python or c++.
For differential action model in Python code, you can find an example here: https://github.com/loco-3d/crocoddyl/blob/devel/examples/notebooks/cartpole_swing_up.ipynb.

For the cost models, you can follow similar approach but you need to inheritance from CostModelAbstract.

jcarpent, 2020-02-13:
> Acceleration cannot be penalize, instead we could penalize velocities (or impact velocities).

You can penalize the acceleration if you want. There is no limit of the formulation in its onw. Maybe of the current API only.

ddliugit, 2020-04-07:
Thanks. I will look into that.

Another issue is about example `quadrupedal_gaits.py` in example folder. 

After running example, this issue happens:

```
jack@ubuntu:~/repos/crocoddyl/examples$ python quadrupedal_gaits.py display
*** SOLVE walking ***
iter     cost         stop          grad          xreg        ureg       step    feas
   0  5.36828e+03  2.29649e+08  2.21345e+06  1.00000e-02  1.00000e-02   1.0000     1
   1  4.03176e+03  4.39596e+05  9.22409e+03  1.00000e-03  1.00000e-03   1.0000     1
   2  3.23235e+03  9.52037e+04  6.81639e+03  1.00000e-03  1.00000e-03   0.5000     1
   3  1.66525e+03  6.81262e+04  5.21547e+03  1.00000e-03  1.00000e-03   0.5000     1
Traceback (most recent call last):
  File "quadrupedal_gaits.py", line 126, in <module>
    ddp[i].solve(xs, us, 100, False, 0.1)
  File "/opt/openrobots/lib/python2.7/site-packages/crocoddyl/__init__.py", line 271, in __call__
    self.visualization.displayFromSolver(solver)
  File "/opt/openrobots/lib/python2.7/site-packages/crocoddyl/__init__.py", line 115, in displayFromSolver
    fs = self.getForceTrajectoryFromSolver(solver)
  File "/opt/openrobots/lib/python2.7/site-packages/crocoddyl/__init__.py", line 176, in getForceTrajectoryFromSolver
    if contact.joint == self.robot.model.frames[c.cost.frame].parent:
```

cmastalli, 2020-04-08:
I has already fixed in the devel branch but for some reason we did not include it in the release.

You have two solutions:
  1. Install Crocoddyl from source using the latest devel branch
  2. Remplace the `GepettoViewer` class (as defined in https://github.com/loco-3d/crocoddyl/blob/devel/bindings/python/crocoddyl/__init__.py#L21) into your installed `__init__` file.

cmastalli, 2020-04-08:
I will close this thread since there are various topics here.

For trackability purpose (to other users and developers), I suggest you to open an issue regarding a single topic.

closed 2020-04-08 by cmastalli