loco-3d/crocoddyl #687: Release v1.1.0

cmastalli, 2020-02-11:
Hi @nim65s,

In this release, we have done the following tasks:
 - added few cost functions related to forces and impulses (e.g. friction-cone, com impulse)
 - improved the display tools (friction cone, contact forces and end-effector trajectories)
 - fixed a problem in the printed message by the callback
 - fixed a problem in the box-qp
 - improved the box-ddp
 - added a new solver called box-fddp
 - added extra examples (box-ddp vs box-fddp, taichi task, humanoid manipulation)
 - added a script for automatically updating the log files
 - checked that all examples runs in the CI (for release mode only)
 - improved the quadrupedal and bipedal examples by adding all the constraints
 - improved the efficiency in differential free-forward dynamics
 - added extra setter functions in action models
 - improved efficiency in all solvers (removed extra computation)
 - improved plot functions
 - fixed a bug in few notebooks

If you like this approach, I will continue using issues for future release.
With it, I could easily describe the improvements / tasks for each release.

Please close this issue whenever you have made the release.

wxmerkt, 2020-02-11:
Can we please use tags rather than branches for releases? Otherwise it will get cluttered. After you create a tag, you can make a new release in the GitHub interface and add the comments there rather than in an issue.

Here's how:
1. Create a tag and push it to GitHub
2. Go to releases -> tags. Click on "x newer tags". Next to the tag there is a ".." dropdown. Click on "Create a release". You can then create a release from that tag and add information, similar to:

[image]

cmastalli, 2020-02-11:
This branch is used by @nim65s to create the tags and release the code on robotpkg.
Certainly, I could handle the tag by myself, but it is better that one person deals with everything.

Note that this branch will be deleted once the release is done. This is how it is also done in Pinocchio.

wxmerkt, 2020-02-11:
Okay, thank you for the explanation. I was under the impression that the `make release` would take care of `git tag` etc.

closed 2020-02-14 by nim65s

cmastalli, 2020-02-14:
Thanks @nim65s :)