loco-3d/crocoddyl #688: float division by zero

ddliugit, 2020-02-11:
I have an error while trying to solve a ddp program. My model is based on cart pole example.

```
class DifferentialActionModelCartpole(crocoddyl.DifferentialActionModelAbstract):
    def __init__(self):
        crocoddyl.DifferentialActionModelAbstract.__init__(self, crocoddyl.StateVector(6), 3, 11) # nu = 1; nr = 6
        self.unone = np.zeros(self.nu)

        self.m = 3.7
        self.g = 9.81
        self.r1 = 0.25
        self.r2 = 0.34
        self.costWeights = [1., 1., 1., 1., 0.1, 0.001, 0.001, 0.001, 1., 1., 1.] 
        # theta-theta_f, phi-phi_f, r-r_f, r*dtheta,
        # tanh(r**2-(r1+r2)*r+r1*r2), dr, dtheta, dphi, f, tau_theta, tau_phi
        
    def calc(self, data, x, u=None):
        if u is None: u=model.unone
        # Getting the state and control variables
        r, dr, theta, dtheta, phi, dphi = np.asscalar(x[0]), np.asscalar(x[1]), np.asscalar(x[2]), np.asscalar(x[3]), np.asscalar(x[4]), np.asscalar(x[5])
        print(r,dr, theta, dtheta, phi, dphi)
        f,tau_theta,tau_phi = np.asscalar(u[0]), np.asscalar(u[1]), np.asscalar(u[2])

        # Shortname for system parameters
        m, g, r1, r2 = self.m, self.g, self.r1, self.r2
        
        # Defining the equation of motions
        # ddr ddtheta ddphi
        ddr  = (f/m+r*dtheta**2+r*dphi**2-g*np.cos(theta))
        print(m)
        print(m*r**2)
        print(r)

        ddtheta = (tau_theta/(m*r**2)+g*np.sin(theta)/r-2*dr*dtheta/r)
        ddphi = (tau_phi/(m*r**2)-2*dr*dphi/r) 
        
        data.xout = np.matrix([ ddr, ddtheta, ddphi]).T

        # Computing the cost residual and value
        data.r = np.matrix(self.costWeights * np.array([theta-theta_f, phi-phi_f, r-r_f, r*dtheta, np.tanh(r**2 - (r1 + r2)*r + r1*r2), dr, dtheta, dphi, f, tau_theta, tau_phi])).T
        data.cost = .5* np.asscalar(sum(np.asarray(data.r)**2))
    
    def calcDiff(self,data,x,u=None,recalc=True):
        # Advance user might implement the derivatives
        pass
```
I print out one variable `r` . It is zero at the beginning and I think that's why it can rise this error. Different from cart pole, there is one variable `r` which is at the denominator of the equation in my model.
```
Traceback (most recent call last):
  File "simple.py", line 211, in <module>
    done = ddp.solve([], [], 1000)
  File "simple.py", line 119, in calc
    ddtheta = (tau_theta/(m*r**2)+g*np.sin(theta)/r-2*dr*dtheta/r)
ZeroDivisionError: float division by zero
```

Is there any way to solve this problem?

cmastalli, 2020-02-12:
In the first iteration r is equals to zero because you are warm-starting the solver with zeros. Thus, you could improve the warm-start.

Your EoMs presents a singularity point in r=0. A common approach to handle it, it will be by applying an tiny regulation (or like damping in the pseudo inverse). Therefore rnew=r+mu with mu as the tiny regularisation (e.g. 1e-9).

As side comment, I do not understand your EoMs. It seems that your confusing r with r1 and/or r2, isn't it? I will suggest you to double check your equations

cmastalli, 2020-02-12:
I will close this issue since it is not related with Crocoddyl.
Feel free to ask more questions, I am happy to help you.

closed 2020-02-12 by cmastalli

ddliugit, 2020-02-12:
we want to bound r within r1 and r2 and right now we are using activation function `tanh`, but it doesn't work well. Is there other method to add bounding box constraint for a variable in a simple customized mode like cartpole?

cmastalli, 2020-02-12:
Yes, it is possible through quadratic barrier. However, this is easy to integrate for our action models, you will need to handle by yourself.

ddliugit, 2020-02-13:
Yes, you are right. But I am not familiar with crocoddyl structure. 

Can you give some hints or an example which I can follow?

I tried to add a constraint like following in the `DifferentialActionModelCartpole`, but it doesn't work well.
```
        ddr  = (f/m+r*dtheta**2+r*dphi**2-g*np.cos(theta))
        if r < 1e-9:
            r+=1e-8
        if r < r1: r = r1
        elif r > r2: r = r2
        print(r)
        
        ddtheta = (g*np.sin(theta)/r-2*dr*dtheta/r)
        ddphi = (-2*dr*dphi/r) 
```

Can you help me?

cmastalli, 2020-02-13:
1. You create your cost model using `CostModelAbstract` (e.g. https://github.com/loco-3d/crocoddyl/blob/devel/bindings/python/crocoddyl/utils/__init__.py#L382)
2. You use the `ActivationModelQuadraticBarrier` as your activation model.
3. You add the data produced by this cost in each `calc` and `calcDiff` function.

If you have checked our notebooks, then you will have a clear idea about point 3.

ddliugit, 2020-02-14:
First, I just copied the code to my code and rename the class.
```
class CostModelBound(crocoddyl.CostModelAbstract):
    def __init__(self, state, activation=None, xref=None, nu=None):
        activation = activation if activation is not None else crocoddyl.ActivationModelQuad(state.ndx)
        self.xref = xref if xref is not None else state.zero()
        crocoddyl.CostModelAbstract.__init__(self, state, activation, nu)

    def calc(self, data, x, u):
        data.r = self.state.diff(self.xref, x)
        self.activation.calc(data.activation, data.r)
        data.cost = data.activation.a

    def calcDiff(self, data, x, u, recalc=True):
        if recalc:
            self.calc(data, x, u)
        data.Rx = self.state.Jdiff(self.xref, x, 'second')[0]
        self.activation.calcDiff(data.activation, data.r, recalc)
        data.Lx = data.Rx.T * data.activation.Ar
        data.Lxx = data.Rx.T * data.activation.Arr * data.Rx
        
```
Second, I just changed it to follow the following code.

```
boundConstraint = crocoddyl.CostModelBound(
                self.state, crocoddyl.ActivationModelQuadraticBarrier(crocoddyl.ActivationBounds(cone.lb, cone.ub)), nu = self.actuation.nu)
```

For `calc` and `calcDiff`, I still have no clue after going through notebook. 

In your examples, there are two totally different examples: one is for customized example, like cart pole. The other is Urdf model and all the costs are constructed with specific API.

My example need to mix these two method together?

I have  self-defined cost in the `calc` and don't know how to add the self-defined cost function `CostModelBound` to the library?

ddliugit, 2020-02-15:
In cart pole example, I cannot create a `state` as the input of `CostModelAbstract `. How to do the following work without `state`?

ddliugit, 2020-02-15:
About the quadratic barrier, I only know the following way to do that, is there a other way to integrate with the mode?

```
        costModel = crocoddyl.CostModelSum(self.state, self.actuation.nu)
        if isinstance(comTask, np.ndarray):
            comTrack = crocoddyl.CostModelCoMPosition(self.state, comTask, self.actuation.nu)
            costModel.addCost("comTrack", comTrack, 1e6)
        for i in supportFootIds:
            cone = crocoddyl.FrictionCone(self.nsurf, self.mu, 4, False)
            frictionCone = crocoddyl.CostModelContactFrictionCone(
                self.state, crocoddyl.ActivationModelQuadraticBarrier(crocoddyl.ActivationBounds(cone.lb, cone.ub)),
                cone, i, self.actuation.nu)
            costModel.addCost(self.rmodel.frames[i].name + "_frictionCone", frictionCone, 1e1)
        if swingFootTask is not None:
            for i in swingFootTask:
                xref = crocoddyl.FrameTranslation(i.frame, i.oMf.translation)
                footTrack = crocoddyl.CostModelFrameTranslation(self.state, xref, self.actuation.nu)
                costModel.addCost(self.rmodel.frames[i.frame].name + "_footTrack", footTrack, 1e6)

        stateWeights = np.array([0.] * 3 + [500.] * 3 + [0.01] * (self.rmodel.nv - 6) + [10.] * 6 + [1.] *
                                (self.rmodel.nv - 6))
        stateReg = crocoddyl.CostModelState(self.state,
                                            crocoddyl.ActivationModelWeightedQuad(np.matrix(stateWeights**2).T),
                                            self.rmodel.defaultState, self.actuation.nu)
        ctrlReg = crocoddyl.CostModelControl(self.state, self.actuation.nu)
        costModel.addCost("stateReg", stateReg, 1e1)
        costModel.addCost("ctrlReg", ctrlReg, 1e-1)

        lb = self.state.diff(0 * self.state.lb, self.state.lb)
        ub = self.state.diff(0 * self.state.ub, self.state.ub)
        stateBounds = crocoddyl.CostModelState(
            self.state, crocoddyl.ActivationModelQuadraticBarrier(crocoddyl.ActivationBounds(lb, ub)),
            0 * self.rmodel.defaultState, self.actuation.nu)
        costModel.addCost("stateBounds", stateBounds, 1e3)
```

ddliugit, 2020-02-15:
```  
File "simple.py", line 222, in <module>
    cartpoleDAM = DifferentialActionModelCartpole()
  File "simple.py", line 185, in __init__
    boundConstraint = CostModelBound(self.state, crocoddyl.ActivationModelQuadraticBarrier(crocoddyl.ActivationBounds(lb, ub,0.9)))
  File "simple.py", line 17, in __init__
    crocoddyl.CostModelAbstract.__init__(self, state, activation, nu)
Boost.Python.ArgumentError: Python argument types in
    CostModelAbstract.__init__(CostModelBound, StateVector, ActivationModelQuadraticBarrier, NoneType)
did not match C++ signature:
    __init__(_object* self, boost::shared_ptr<crocoddyl::StateMultibody> state, int nr)
    __init__(_object* self, boost::shared_ptr<crocoddyl::StateMultibody> state, int nr, bool withResiduals)
    __init__(_object* self, boost::shared_ptr<crocoddyl::StateMultibody> state, int nr, int nu)
    __init__(_object* self, boost::shared_ptr<crocoddyl::StateMultibody> state, int nr, int nu, bool withResiduals)
    __init__(_object* self, boost::shared_ptr<crocoddyl::StateMultibody> state, boost::shared_ptr<crocoddyl::ActivationModelAbstract> activation)
    __init__(_object* self, boost::shared_ptr<crocoddyl::StateMultibody> state, boost::shared_ptr<crocoddyl::ActivationModelAbstract> activation, bool withResiduals)
    __init__(_object* self, boost::shared_ptr<crocoddyl::StateMultibody> state, boost::shared_ptr<crocoddyl::ActivationModelAbstract> activation, int nu)
    __init__(_object* self, boost::shared_ptr<crocoddyl::StateMultibody> state, boost::shared_ptr<crocoddyl::ActivationModelAbstract> activation, int nu, bool withResiduals)
server@server:~/landing/src/crocoddyl$ python simple.py 
/home/server/.local/lib/python2.7/site-packages/pkg_resources/py2_warn.py:22: UserWarning: Setuptools will stop working on Python 2
************************************************************
You are running Setuptools on Python 2, which is no longer
supported and
>>> SETUPTOOLS WILL STOP WORKING <<<
in a subsequent release (no sooner than 2020-04-20).
Please ensure you are installing
Setuptools using pip 9.x or later or pin to `setuptools<45`
in your environment.
If you have done those things and are still encountering
this message, please comment in
https://github.com/pypa/setuptools/issues/1458
about the steps that led to this unsupported combination.
************************************************************
  sys.version_info < (3,) and warnings.warn(pre + "*" * 60 + msg + "*" * 60)
/home/server/.local/lib/python2.7/site-packages/robot_properties_solo/config/package_config.ini
/home/server/repos/humanoid_property/urdf/biped_new.urdf
['/home/server/repos/humanoid_property']
eigenpy warning: you use the deprecated class numpy.matrix without explicily asking for it. The default behaviour will change to numpy.array at next major release.
- Either call eigenpy.switchToNumpyMatrix() before using eigenpy to suppress this warning
- or call eigenpy.switchToNumpyArray() and adapt your code accordingly.
See https://github.com/stack-of-tasks/eigenpy/issues/87 for further details.
0.15
-2.42861286637e-17
9.99999998968e-06
6.93889390391e-18
3.46944695195e-18
9.99999998971e-06
(-0.05899000000000883, 0.10125000000000334, 2.5729418595688003e-14)
(9.999999989712682e-06, 0.0612500000000012, 2.5777990853015353e-14)
[0.007158208953623851, -5.05987806359964e-06, 0.3012260236425874]
[ 0.06614821 -0.10125506  0.30122602]
(0.32460020027441988, 21.876236037253879, 303.15590254307097)
(-2.0166440755912443, -1.2652270590217869, 1.2579138132248224)
(0.32460020027441988, 78.24169649626711, 108.57971727347046)
(matrix([[ 0.25],
        [ 0.  ],
        [ 0.  ],
        [ 0.  ],
        [ 0.  ],
        [ 0.  ],
        [ 0.  ],
        [ 0.  ]]), matrix([[ 0.34],
        [ 0.  ],
        [ 0.  ],
        [ 0.  ],
        [ 0.  ],
        [ 0.  ],
        [ 0.  ],
        [ 0.  ]]))
Traceback (most recent call last):
  File "simple.py", line 222, in <module>
    cartpoleDAM = DifferentialActionModelCartpole()
  File "simple.py", line 185, in __init__
    boundConstraint = CostModelBound(self.state, crocoddyl.ActivationModelQuadraticBarrier(crocoddyl.ActivationBounds(lb, ub,0.9)),2)
  File "simple.py", line 17, in __init__
    crocoddyl.CostModelAbstract.__init__(self, state, activation, nu)
Boost.Python.ArgumentError: Python argument types in
    CostModelAbstract.__init__(CostModelBound, StateVector, ActivationModelQuadraticBarrier, NoneType)
did not match C++ signature:
    __init__(_object* self, boost::shared_ptr<crocoddyl::StateMultibody> state, int nr)
    __init__(_object* self, boost::shared_ptr<crocoddyl::StateMultibody> state, int nr, bool withResiduals)
    __init__(_object* self, boost::shared_ptr<crocoddyl::StateMultibody> state, int nr, int nu)
    __init__(_object* self, boost::shared_ptr<crocoddyl::StateMultibody> state, int nr, int nu, bool withResiduals)
    __init__(_object* self, boost::shared_ptr<crocoddyl::StateMultibody> state, boost::shared_ptr<crocoddyl::ActivationModelAbstract> activation)
    __init__(_object* self, boost::shared_ptr<crocoddyl::StateMultibody> state, boost::shared_ptr<crocoddyl::ActivationModelAbstract> activation, bool withResiduals)
    __init__(_object* self, boost::shared_ptr<crocoddyl::StateMultibody> state, boost::shared_ptr<crocoddyl::ActivationModelAbstract> activation, int nu)
    __init__(_object* self, boost::shared_ptr<crocoddyl::StateMultibody> state, boost::shared_ptr<crocoddyl::ActivationModelAbstract> activation, int nu, bool withResiduals)
```

ddliugit, 2020-02-15:
The point 1 you mentioned cannot be applied to the cart pole example, can you help to solve this urgent issue?

cmastalli, 2020-02-17:
At the time being, this class is made for robotics problem. I am afraid you will need to implement yourself the quadratic barrier function.

ddliugit, 2020-02-17:
Is it easy to do that? Is there  a simplified way? Do you have the code I can followv

cmastalli, 2020-02-17:
Yes, I would say so.
I don't have any code to provide you. You must want to see the c++ implementation of the quadratic barrier class.

ddliugit, 2020-02-17:
Would that be possible if I encode the following into the cost function?

-mu*log[(y1-y)(y-y2)] and mu is a small number, say 1e-5