loco-3d/crocoddyl #691: Code restructuring.

proyan, 2020-02-17:
@wxmerkt @PepMS @VladimirIvan @nim65s @cmastalli @MaximilienNaveau 

In order to use CodeGen on crocoddyl ActionModels, I'm restructuring the code on the `Scalar` template. This PR would be available in this week.
There would be no change in API, and it wouldn't affect your dependent code. 
Nevertheless, I opened this issue so you could let me know if you see any way this might affect your ongoing work, and if so, how we could plan the merge schedule.

cmastalli, 2020-02-17:
Please do not modify anything that is not in the scope of this task.
I am trying to prevent much bigger PRs.

Consider that there is 20% of the code that it is not currently tested in our CI.

proyan, 2020-03-02:
Fixed with #695

closed 2020-03-02 by proyan