loco-3d/crocoddyl #693: Assertion from pinocchio explog triggered in examples

wxmerkt, 2020-02-17:
I am on 18.04 and  have the latest Pinocchio devel compiled with `CMAKE_BUILD_TYPE=MinSizeRel -Os` and Crocoddyl compiled with `CMAKE_BUILD_TYPE=RelWIthDebInfo -O3`. Running any of the examples (e.g. `boxddp_vs_boxfddp` or `quadrupedal_walk_ubound`) results in the following assertion:

```
include/pinocchio/spatial/explog.hpp:94: Eigen::Matrix<typename Matrix3Like::Scalar, 3, 1, typename Eigen::internal::plain_matrix_type<typename pinocchio::helper::argument_type<void(D)>::type, typename Eigen::internal::traits<typename pinocchio::helper::argument_type<void(D)>::type>::StorageKind>::type:: Options> pinocchio::log3(const Eigen::MatrixBase<Derived>&, typename Matrix3Like::Scalar&) [with Matrix3Like = Eigen::Matrix<double, 3, 3>; typename Eigen::internal::plain_matrix_type<typename pinocchio::helper::argument_type<void(D)>::type, typename Eigen::internal::traits<typename pinocchio::helper::argument_type<void(D)>::type>::StorageKind>::type = Eigen::Matrix<double, 3, 3>; typename Matrix3Like::Scalar = double]: Assertion `theta == theta && "theta contains some NaN"' failed.
```

This is on an Intel(R) Core(TM) i7-9850H CPU @ 2.60GHz which has been freshly installed. I did not see these issues on my previous configurations (i7-8565U or i9-9900K) - albeit with older commits for both Pinocchio and Crocoddyl back then. 

Have you seen these before or any insights why these might be triggered?

wxmerkt, 2020-02-17:
I cannot reproduce this on another machine (i7-8565U) that should be set up in the same way. Any ideas why this could happen though? I saw it a few times in the past. Clean builds did not resolve it.

cmastalli, 2020-02-18:
I found the problem. I will make a PR that solves it.
You will see more details there.

cmastalli, 2020-02-18:
Please see the PR.

wxmerkt, 2020-02-18:
Thank you for the quick turn-around and fix :-) I did not expect it to be in the `quadruped.py`.

cmastalli, 2020-02-18:
> Thank you for the quick turn-around and fix :-) I did not expect it to be in the `quadruped.py`.

You're welcome, and thank you for the report.
Unfortunately, we are not testing the examples in Debug mode because it takes few minutes to run them. Did you have any suggestion on the topic?

closed 2020-02-18 by cmastalli

wxmerkt, 2020-02-19:
I think having them run on CI even if it takes minutes is fine. Pull requests shouldn't be merged that quickly for the jobs not to complete. We have a 50 minute limit on the jobs and are currently at <20min

cmastalli, 2020-02-19:
> I think having them run on CI even if it takes minutes is fine. Pull requests shouldn't be merged that quickly for the jobs not to complete. We have a 50 minute limit on the jobs and are currently at <20min.
> 
> The bigger issue is that we also need Pinocchio and EigenPy in Debug for that. This wasn't Crocoddyl debug that triggered it but Pinocchio in this case

We could include the examples in Debug mode, not problem at all.

Are you sure about your last statement? As far as I understand, we use Pinocchio header files in Crocoddyl CI jobs. Thus, this problem should be triggered as well in Travis, right?

wxmerkt, 2020-02-19:
> Are you sure about your last statement? As far as I understand, we use Pinocchio header files in Crocoddyl CI jobs. Thus, this problem should be triggered as well in Travis, right?

This is only partially true (for Crocoddyl itself). The Python bindings link [the original argument for Boost.Python to utilise the Pinocchio bindings] and thus it's dependant on how Pinocchio was compiled