loco-3d/crocoddyl #697: Q: How are joint limits considered?

julesser, 2020-02-24:
Hi all,
I generally would like to know how joint limits (e.g. effort limit), specified in the URDF, are considered in crocoddyl. The specific questions are:
- Are the limits integrated into the cost functions and therefore optimized, or somehow set as fixed boundaries for the solver? 
- How is the solution currently implemented?

wxmerkt, 2020-02-24:
a) Control limits can be handled by `BoxDDP` and `BoxFDDP` explicitly
b) State limits (position, velocity) are handled as cost terms/penalisations

cmastalli, 2020-02-24:
`BoxDDP` is similar to Tassa method (https://homes.cs.washington.edu/~todorov/papers/TassaICRA14.pdf), with the improvement that you could warm-start the state trajectory.

Instead `BoxFDDP` is a novel solver that its contribution is under-review (we can't share with you the paper now). This solver is superior to the `BoxDDP`, and you can check it with the above example.

The control limits are handled it automatically by parsing the URDF. However, you could redefine it inside the Pinocchio model.

See this example https://github.com/loco-3d/crocoddyl/blob/devel/examples/boxfddp_vs_boxddp.py#L20

julesser, 2020-02-24:
Many thanks for the quick answers. As expected:

> The control limits are handled it automatically by parsing the URDF. However, you could redefine it inside the Pinocchio model.

Then in turn, I don't quite understand how the control inputs, for example in the Manipulator example, can exceed the limits of the URDF for the Talos arm so much:

[image]

cmastalli, 2020-02-24:
The arm-manipulation example uses `DDP` solver, thus we are not adding constraints to this problem.

Furthermore, it is fairly simple optimal control formulation, without any joint limits.

All examples with '_ubound` are implementing torque limits.

julesser, 2020-02-24:
Got it. Thanks for your comprehensible explanation.

closed 2020-02-24 by julesser