loco-3d/crocoddyl #699: Displaying Robot Poses without Solving the OC

julesser, 2020-02-26:
Hi all,
I am wondering why robots do not seem to be displayed correctly in the gepetto viewer until the `.solve()` of the according shooting problem is called.

**Consider e.g. the bipedal_walking.ipynb**
Executing the script until `.solve()` gives: 
[image]
After calling `.solve()` the pose is shown correctly:
[image]

I came across this issue during the integration of a robot model and was irritated since the information about the reference pose is correctly parsed from the URDF.
The **related questions**:

1. Is there an existing workaround for just displaying a robot in an initial pose without solving the OC?
2. What happens in the `.solve()` method that actually effects the display?
    (thought so far they were decoupled)

cmastalli, 2020-02-26:
> Is there an existing workaround for just displaying a robot in an initial pose without solving the OC?

Yes, it is possible. There are two ways:
  1. Directly interacting with gepetto-viewer, something like this:
```python
crocoddyl_display.robot.display(q0)
```
 2. Using the crocoddyl display (see https://github.com/loco-3d/crocoddyl/blob/devel/bindings/python/crocoddyl/__init__.py#L74):
```python
crocoddyl_display.display(xs=[q0])
```

---
>  What happens in the `.solve()` method that actually effects the display?
>   (thought so far they were decoupled)

Solver runs any callback function that it has been defined. When you active the display in the examples, then we include a callback function that display the robot in the GV at a defined iteration frequency. See this https://github.com/loco-3d/crocoddyl/blob/devel/examples/bipedal_walk.py#L59.

julesser, 2020-02-26:
> crocoddyl_display.display(xs=[x0])

That was exactly what I was searching for - works very well.
Thanks for your explanation.

closed 2020-02-26 by julesser

cmastalli, 2020-02-26:
You're welcome :)