loco-3d/crocoddyl #701: Hybrid simplified model

ddliugit, 2020-02-29:
Hi, folks!

I am wondering if it is possible to solve a hybrid simplified model (i.e. spring loaded inverted pendulum) problem with crocoddyl.

The problem is constructed as two models with different variables.
For the flight phase, we are using x_c, y_c, theta to represent the model.
While it is contacted with ground, we need to use polar coordinate to represent the model.

I am following the cart pole model. If possible, how to connect them?
As it is not Pinocchio based robot problem, I cannot use Pinocchio based API.

cmastalli, 2020-03-02:
Hi @ddliugit 

The API is flexible enough to implement any model you want, including the cart-pole model.
Some of the cost functions are related to Pinocchio, e.g. angular momentum. And of course, you can't use them without Pinocchio model.

You can also use all the activation functions, which are an important component to build a cost function.

Carlos

ddliugit, 2020-03-02:
What is the difference between `ActivationModelQuadraticBarrier` and `ActivationModelInequality`?
It seems both can control the variable between `lowerlimit` and `upperlimit`.

cmastalli, 2020-03-03:
`ActivationModelInequality` has been renamed to `ActivationModelInequality` with futher improvements in the API.

Note that `ActivationModelInequality` is deprecated, as any class inside unittest/python/crocoddyl.

cmastalli, 2020-03-18:
I will close this thread since there is no issue on the code.
Feel free to keep asking if you need further help.

closed 2020-03-18 by cmastalli