loco-3d/crocoddyl #702: Gepetto display doesn't show the motion animation

gfadini, 2020-03-02:
Hello, I'm using the file ```arm_manipulation.py``` trying to show the motion of the arm manipulator,
however it seems to not be working for the ddp solver anymore
This:
```if WITHDISPLAY:
    display = crocoddyl.GepettoDisplay(talos_arm, 4, 4, cameraTF)
    display.displayFromSolver(ddp)
```
doesn't produce an animation
Moreover also with the fddp, GepettoDisplay doesn't show the overall motion at the end, just the callbacks.
I think this may be an easy bug to solve, @cmastalli how was the function conceived?

cmastalli, 2020-03-02:
Hi @gfadini 

It works just the motion is very fast because you're asking the viewer to sample it. Do as follows:
```python
display = crocoddyl.GepettoDisplay(talos_arm)
display.displayFromSolver(ddp)
```

I will close this issue since there is no any bug.
Feel free to ask more question in this thread.

closed 2020-03-02 by cmastalli

gfadini, 2020-03-02:
Ok, got it, thanks