loco-3d/crocoddyl #703: Model multiple point-contacts per foot

julesser, 2020-03-04:
Hi all, 
I am trying to model multiple point-contacts per foot in the context of bipedal walking. Currently it seems shooting problem can not be solved properly (plotConvergence graphics stay all empty, Robot does not move at all).
To this end, I've modified the **biped_utils** as follows: 
1. Pass the main frame (for foot tracking) and the contact frames (4 additional per foot) to the utils:
`def __init__(self, rmodel, rightFoot, leftFoot, rightFootContacts, leftFootContacts)`
2. Modify the walking problem to consider the contact frames for the supportingFoot and the main frame for the swingingFoot, e.g.
`rStep = self.createFootstepModels(comRef, [rfPos0], 0.5 * stepLength, stepHeight, timeStep, stepKnots, self.lfContactIds, [self.rfId])`
3. Don't touch createSwingFootModel(), which e.g. is supposed to yield for a step node in:
   - a contact model containing 8 contact items (4 for each foot)
   - a cost model containing one CoM tracking item, one foot tracking item, 8 friction cone items, state/ctrl reg items

The questions:
- Are the solvers generally able to handle multiple point-contacts that lay on the same planar surface of one foot?
-  Any other ideas what could cause the problems?

cmastalli, 2020-03-04:
**About your developed method**
Adding 4 per each foot is mathematically wrong within this context.
Each point-contact introduces a 3d constraint Jacobian, i.e. 12 constraints.
Furthermore, for the case of the double support phase, you will have a total of 24 constraints but the system DoFs is 18.
In conclusion, you are over-constraining the dynamical systems.
See the equations of this paper for more details: https://cmastalli.github.io/publications/crocoddyl20icra.pdf

**About the example model**
As you see, I encode a 6d Jacobian, which constraints rotation too.
The only missing part of this model is the constraint that keeps the CoP inside the foot support.
Unfortunately, this cost function remains to be developed. Is this feature important for you now?

I would definitively develop this cost function, but it is not yet in the TOP of my TODO list.
I could commit to it if this is important, or you are also free to make this contribution.

julesser, 2020-03-06:
> In conclusion, you are over-constraining the dynamical systems.

Thanks for pointing this out. Actually I thought these redundant constraints could be handled by the dynamic solvers. For now,  we will simply stick with modelling one point contact per foot. 

> Is this feature important for you now?

No, currently this is not crucial for us.

closed 2020-03-06 by julesser