loco-3d/crocoddyl #704: Pickling issues

gfadini, 2020-03-04:
Hello,
can crocoddyl be pickled yet? I rember that @cmastalli said that there were issues with eigen for that, but maybe I'm mistaken and that is not the case anymore .
I have a runtime error related to this using multiprocessing, it would be great if it could be fixed/was supported.

cmastalli, 2020-03-04:
You cannot pickle entire Crocoddyl data. However you could pickle some data, e.g. Pinocchio data, NumPy, etc.

Unfortunately I don't have time now for adding this feature.

cmastalli, 2020-04-02:
I would like to take this thread to discuss which classes / structures should be pickleable. I was thinking in this set of classes / structures:
 1. All data structures
 2. All model classes
 3. All solver classes
 4. `Python list` to/from `std::vector` 
 4. `Python Dict` to/from `std::map`
 
What do you think? @proyan @gfadini @jcarpent @wxmerkt 

---

This also requires a lot of work. I notice that there is a light alternative that will be faster to deploy but we need to inject a `__getinitargs__` method, see details:
https://www.boost.org/doc/libs/1_58_0/libs/python/doc/v2/pickle.html (picke4)
is this would make sense for you guys?

jcarpent, 2020-04-02:
@cmastalli You should take inspiration from https://github.com/stack-of-tasks/pinocchio which is using the serialization for the pickling of large data structure, which is in the end much more efficient and allows to export in C++ too.

jcarpent, 2020-04-02:
The solver should not be pickable at first glance. Only data and model in the first place.
By the way, for Python list, you should use the code from Pinocchio too.

cmastalli, 2020-07-23:
> @cmastalli You should take inspiration from https://github.com/stack-of-tasks/pinocchio which is using the serialization for the pickling of large data structure, which is in the end much more efficient and allows to export in C++ too.

I reviewed again this code, and I have two questions.
 1. It seems to me that `getinitargs` assumes that there are no arguments for the object construction. So if I understand correctly, this function can be removed, right?
 2. Does this code assume that the state of the instance is always a public member?

Ideally I would like to reuse this code for pickling action models and datas, however, it seems to me that 2 provides its use. I mean, I need to set some private states using `set_XXX` functions.

For me, the serialization mechanism is a bit obscure.