loco-3d/crocoddyl #713: Bipedal walking: CoM decreases with every gait phase

julesser, 2020-03-12:
Hi all, 
for the bipedal walking of talos there emerges an inperiodicity in the joint trajectories (knee, ankle joints) during a 6-step gait.
[image]
This leads to a decresaing CoM with every gait phase. 
I've tried 

- Increasing the weight for the CoM cost item and 
- Increasing the weight of the state regression cost item

without success so far. 

1.) Any ideas on how to prevent the CoM decreasing?
2.) Is there a cost function to penalize differences between initial and terminal pose of the robot?

cmastalli, 2020-03-12:
The optimal control (OC) formulation needs to be improved, and I am working on it.

With the current formulation, we build a CoM reference (for cost tracking) that starts in the initial position on the robot. However, the relative CoM positions w.r.t the foot placements varies from each OC problem. It leads with different OC problems and solutions

As mentioned in another thread, the current OC formulation is fairly simple and not done for robot deployment (only for examples). I will push a need formulation asap (it will take few weeks, unfortunatly).

julesser, 2020-03-12:
Thanks for your quick answer. We are looking forward to progress in this direction :-)

cmastalli, 2020-03-12:
Feel free to ask any further question if it is needed.

I will close this issue.

closed 2020-03-12 by cmastalli