loco-3d/crocoddyl #728: Bad initialization of dx0 in the forward pass

edantec, 2020-03-27:
The gap f0 should be used to initialize the first dx0, rather than force it to be dx0 = 0. In practice, f0 should be equal to the difference of the initial constrained state x0 (provided during the creation of the shooting problem) and the first state of the warm state trajectory, provided during the ddp->solve.

cmastalli, 2020-03-27:
@edantec -- Nicolas explained the solution, could you provide a PR with this work?
If so, note that you will need to review:
  1. `SolverDDP`
  2. `SolverFDDP`
  3. `SolverKKT`
and check that the unit-tests are passing.

cmastalli, 2020-03-30:
@edantec, @proyan and @nmansard 

I review a second time the code for both: DDP and FDDP, and **I conclude that we don't have this issue. I mean the solvers are properly handling the initial condition**. Let me to explain it as easy as possible.

However, before starting the explanation, we need to say that `dx` has `T` length (not `T+1` length). In practice means that `dx0` isn't associated to the initial condition, instead it's associated to the first equality condition (dynamic condition). Note that we don't need to allocate another `dx` for the initial condition in any of the solvers.

**DDP solver***
In this case, we already know that the first forward pass will close the gaps. To ensure this, I have defined the `xs_try` with the initial state (https://github.com/loco-3d/crocoddyl/blob/devel/src/core/solvers/ddp.cpp#L43). And this is needed for running the forward pass (i.e. `dx0 = x0 - xs[0]`) as you can see in https://github.com/loco-3d/crocoddyl/blob/devel/src/core/solvers/ddp.cpp#L250.

***FDDP solver**
If the user provides an infeasible warm-start for the initial condition, then the FDDP solver accounts it during the calculation of the gap `f0` (https://github.com/loco-3d/crocoddyl/blob/devel/src/core/solvers/ddp.cpp#L156) (as well the DDP solver). Then the forward pass will close the gaps as much as defined by the step length (https://github.com/loco-3d/crocoddyl/blob/devel/src/core/solvers/box-fddp.cpp#L98).

---
Do you agree with me?

cmastalli, 2020-03-30:
I will close this issue.
Feel free to re-open if that the case.

closed 2020-03-30 by cmastalli