loco-3d/crocoddyl #737: A simple model cannot be optimized

ddliugit, 2020-03-31:
I built a simple example with crocoddyl. There is zero gravity and zero friction environment. a point mass of 1kg that can only be pushed along one direction with a force f. 

I want to optimize the mass to distance 1m with least force f.

The code is like the following:
I got nothing from the result.

```
class ActionModelCar(ActionModelAbstract):
    def __init__(self):
        '''
        Consider in a zero gravity and zero friction environment a point mass of \SI{1}{\kilo\gram} that can only be pushed along one direction with a force f. 
        Denote by $s$ the position of the point mass, we have the following ODE that completely describes the system by Newton's second law of motion

        The transition model of a LIPM with flywheel is described as:
            x = [s ds]
            u = [f]
            s_next = s + ds*dt
            ds_next = ds + f/m*dt
        '''
        ActionModelAbstract.__init__(self, crocoddyl.StateVector(2), 1)
        self.ActionDataType = ActionDataAbstract
        self.dt = 1e-3

        self.ncost = 2
        # weights of x-x_f, f
        self.costWeights = [1.,0.1]
        self.unone = np.zeros(self.nu)
        self.m = 1.0

    def calc(self, data, x, u=None):
        if u is None:
            u = self.unone
        #assert (x.shape == (self.nx, ) and u.shape == (self.nu, ))
        #assert (data.xnext.shape == (self.nx, ))
        #assert (data.costResiduals.shape == (self.ncost, ))
        #assert (len(self.costWeights) == self.ncost)

        y, dy = x
        f = np.asscalar(u)
        dt = self.dt
        m = self.m

        k1, k2 = self.costWeights

        # differential model 
        data.xnext[0] = y + dy*dt
        data.xnext[1] = dy + (f/m)*dt

        # cost 
        cost1 = k1 * (y-1)**2  # q-q_f
        cost2 = k2 * f**2  # f
        
        data.cost = .5 * (cost1+cost2)
        
        return data.xnext, data.cost

    def calcDiff(self, data, x, u=None,recalc=True):
        if u is None:
            u = self.unone
        xnext, cost = self.calc(data, x, u)

        y, dy = x
        f = u
        dt = self.dt
        m = self.m
        k1, k2 = self.costWeights

        # Cost derivatives
        data.Lx[:] = [k1*(y-1),
                      0]  # to be completed, 6*1
        data.Lu[:] = [k2*f]  # to be completed, 2*1
        data.Lxx[:] = [[k1, 0],
                       [0, 0]]  # to be completed, 6*6
        data.Luu[:] = [[k2]]  # to be completed, 2*2
        # data.Lxu[:] = [[0,0],[0,0],[0,0],[0,0],[0,0],[0,0]] # to be completed, 6*2
        # Dynamic derivatives
        data.Fx[:] = [[1, dt],
                      [0, 1]]
        data.Fu[:]=[0,f/m] 
        return xnext, cost

class testCar():
    def __init__(self):
        '''
        This class is to test LIPM model
        '''

        # Initialize initial contion and termial condition
        x0=np.array([0,0])
        self.x0=x0.T
        self.nx=0
        self.nu=0
        self.horizon_length = 2000
        self.timeStep = 1e-3
        self.timeLength = self.horizon_length*self.timeStep

        self.problem = self.createProblem()

        ddp = crocoddyl.SolverDDP(self.problem)
        ddp.setCallbacks([ crocoddyl.CallbackVerbose() ])
        self.done = ddp.solve([], [], 300)
        

        us_g = [ pinocchio.utils.zero(1) ]*self.horizon_length
        xs_g = self.problem.rollout(us_g)
        self.plot(ddp.xs,ddp.us,xs_g,us_g)

        
    def createProblem(self):
        # extract q_f from calculated terminal condition
        CarDAM = ActionModelCar()
        terminalCar = ActionModelCar()
        self.nx = 2
        self.nu = 1
        # construct terminal cost
        # weights of q-q_f, r*dq, theta, dtheta, f, tau, r barrier
        terminalCar.costWeights[0] = 10.
        terminalCar.costWeights[1] = 0.1

        problem = crocoddyl.ShootingProblem(self.x0, [ CarDAM ]*self.horizon_length, terminalCar)

        return problem

    def plot(self,xs,us,xs_g,us_g):
        T=self.horizon_length
        timeStep=self.timeStep
        x=np.zeros([self.nx,T+1])
        u=np.zeros([self.nu,T+1])
        x_g=np.zeros([self.nx,T+1])
        u_g=np.zeros([self.nu,T+1])

        for i in range(T+1):
            for j in range(self.nx):
                x[j,i] = xs[i][j]
                x_g[j,i] = xs_g[i][j]

        for i in range(T):
            for j in range(self.nu):
                u[j,i] = us[i][j]
                u_g[j,i] = us_g[i][j]

        plt.figure(figsize=(18, 12))
        ax1=plt.subplot(3,2,1)
        ax1.set_ylabel('s')
        plt.plot(np.arange(T+1)*timeStep, x[0,:],'g-',linewidth=8)
        plt.plot(np.arange(T+1)*timeStep, x_g[0,:],'b--',linewidth=8)
        ax2=plt.subplot(3,2,2)
        ax2.set_ylabel(r'$\dot{s}$')
        plt.plot(np.arange(T+1)*timeStep, x[1,:],'g-')
        plt.plot(np.arange(T+1)*timeStep, x_g[1,:],'b--')
        ax3=plt.subplot(3,2,3)
        ax3.set_ylabel('f')
        plt.plot(np.arange(T+1)*timeStep, u[0,:],'g-')
        plt.plot(np.arange(T+1)*timeStep, u_g[0,:],'b--')

        plt.show(block=True)
        

def main():
    test = testCar()
    print(test.done)

if __name__ =="__main__":
    main()
```

cmastalli, 2020-04-01:
Hi @ddliugit 

Recently, the Python interface of Crocoddyl allows us to write internal data in NumPy objects such as `data.xnext`. For instance the operation:
```python
data.xnext[0] = # some value to be defined internally
data.xnext[1] = # some value to be defined internally
```
is not allowed, instead you could do (Crocoddyl v1.1.0 or lower)
```python
xnext = # your vector
data.xnext = xnext # copy a new vector
```
However, I suggest you to update Crocoddyl. This stuff will be release soon under v1.2.0.

If you decide to move to v1.2.0, then you need to clean other stuff in your code:
 1. Remove recalc option in `calcDiff`. Now `calcDiff` doesn't run `calc`
 2. Remove Python asserts since you might want to use NumPy array (Crocoddyl internally is running those asserts)
 3. If you want to keep NumPy matrix (slower than array), then you need to add `crocoddyl.switchToNumpyMatrix()`

Let me know if this helps you!

Yours,
Carlos

ddliugit, 2020-04-03:
Thanks, Carlos!

I saw you released v1.2.0 version a few hours ago. Can you update robotpkg package with the latest version. I tried to build from source, but some errors happen.

```
[ 12%] Building CXX object CMakeFiles/crocoddyl.dir/src/multibody/actions/contact-fwddyn.cpp.o
/usr/bin/c++   -DBOOST_MPL_LIMIT_LIST_SIZE=30 -DBOOST_MPL_LIMIT_VECTOR_SIZE=30 -DPINOCCHIO_URDFDOM_TYPEDEF_SHARED_PTR -DPINOCCHIO_WITH_URDFDOM -Dcrocoddyl_EXPORTS -I/home/jack/repos/crocoddyl/build -I/home/jack/repos/crocoddyl/build/include -I/home/jack/repos/crocoddyl/include -I/home/jack/repos/crocoddyl/bindings -isystem /usr/include/eigen3  -pedantic -Wno-long-long -Wall -Wextra -Wcast-align -Wcast-qual -Wformat -Wwrite-strings -Wconversion  -O3 -DNDEBUG -fPIC   -o CMakeFiles/crocoddyl.dir/src/multibody/actions/contact-fwddyn.cpp.o -c /home/jack/repos/crocoddyl/src/multibody/actions/contact-fwddyn.cpp
In file included from /home/jack/repos/crocoddyl/src/multibody/impulse-base.cpp:9:0:
/home/jack/repos/crocoddyl/include/crocoddyl/multibody/impulse-base.hpp:12:40: fatal error: pinocchio/multibody/data.hpp: No such file or directory
compilation terminated.
```

cmastalli, 2020-04-03:
You might have a problem with cmake files. Please remove the cmake files and start from scratch

ddliugit, 2020-04-04:
Currently my code is based on cartpole example you provided the `example` folder and `bipedal_walk.py`,

After updating to v1.2.0, both of them don't work normally.

See the following errors:
```
jack@ubuntu:~/repos/crocoddyl/examples/notebooks$ python cartpole_swing_up.py 
Traceback (most recent call last):
  File "cartpole_swing_up.py", line 86, in <module>
    cartpoleND = DifferentialActionModelNumDiff(cartpoleDAM)
Boost.Python.ArgumentError: Python argument types in
    DifferentialActionModelNumDiff.__init__(DifferentialActionModelNumDiff, instance)
did not match C++ signature:
    __init__(_object* self, boost::shared_ptr<crocoddyl::DifferentialActionModelAbstractTpl<double> > model)
    __init__(_object* self, boost::shared_ptr<crocoddyl::DifferentialActionModelAbstractTpl<double> > model, bool gaussApprox)
```

```
jack@ubuntu:~/repos/crocoddyl/examples$ python bipedal_walk.py 
*** SOLVE walking ***
Traceback (most recent call last):
  File "bipedal_walk.py", line 70, in <module>
    xs = [talos_legs.model.defaultState] * len(ddp[i].models())
AttributeError: 'SolverDDP' object has no attribute 'models'
```

cmastalli, 2020-04-06:
Sorry for the problems in the `cartpole_swing_up` script. I have updated the notebooks, I created a PR that fixes it.

Regarding your own code, you need to retrieve the models through the problem. Basically, you need to replicate this line:
https://github.com/loco-3d/crocoddyl/blob/devel/examples/bipedal_walk.py#L73

ddliugit, 2020-04-06:
Thanks. I replicate the code `ActionModelUnicycle`. Now the carpool works fine!

Do you have any schedule to replicate the `jump.py` in `npydata` folder. Now I am still using the v0.2.0 the pure python version for that optimization. I really hope to restore this example by using the latest API.

cmastalli, 2020-04-06:
I do not have a hard deadline, but I will finish it within the next weeks. In any case, you might be interested in the example about the quadrupedal jump, this is already available.

I will close this issue. Feel free to re-open it!

closed 2020-04-06 by cmastalli