loco-3d/crocoddyl #742: Cannot import name 'ActionModelImpact'

julesser, 2020-04-02:
Hi all, 
It seems to me as if the examples for highly-dynamics movements (jump.py/salto.py) became out-of-date. I wanted to report that currently both files are not executable, because of (at least) the above dependency error.
Best, 
Julian

cmastalli, 2020-04-02:
Yes, they are backup files. We'll soon removed them!

These examples use an old API interface, based on pure Python code.

julesser, 2020-04-02:
Okay, thanks for the comment. Will also a new version for these maneuvers be uploaded again, matching the new API interface?

cmastalli, 2020-04-02:
> Okay, thanks for the comment. Will also a new version for these maneuvers be uploaded again, matching the new API interface?

At the moment, I have not the resource to do it. But I will attend to do it with a new formulation! Therefore, it won't look like those example codes.

julesser, 2020-04-02:
Okay, thanks for the clarification.

closed 2020-04-02 by julesser