loco-3d/crocoddyl #743: Numerical impact of changes to BoxQP

wxmerkt, 2020-04-02:
In https://github.com/loco-3d/crocoddyl/pull/658, the BoxQP was changed substantially.

From the PR description:
> One of the main problem was due to a bug in the box QP, incorrect output of the active set. 

Can you please elaborate and provide an example / line numbers from the old solver?

> Additionally, our box QP is able to accept infeasible warm-start in contrast to the original proposed method by Tassa, this is done by a "cheap" projection to the feasible set. 

Correct, this step was omitted and is new.

> In conclusion, this PR proposes a completely new box QP solver, which is based on Cholesky factorization.

The old solver relied on the general inverse (`.inverse()`), which in our benchmarks wasn't too much of an overhead. In fact, with compiler optimisations I sometimes get opposite results (faster convergence using the old).

Unfortunately, no test cases or any further detailed description was included.

From a side-by-side inspection, two notable changes to the originally introduced BoxQP exist:

1) Change of the L2 norm to the infinity norm for gradient tolerance check. This has no influence on convergence or numerical behaviour (verified just now).
2) Change from a linear, backtracking line search (alpha=1.0, 0.9, ... 0.1) to an exponential line search. This has a large numerical impact (see below).

Additionally, I believe the analysis that there was a bug in the active set is correct. I have not been able to pinpoint it down _where_ yet. 

To figure out the different numerical behaviours, I am testing all different combinations on a quadcopter task (in Exotica "example_dynamic_time_indexed 07_..."). The only element that is being changed are the changes to the BoxQP.

[image]

This plot shows:

1. The original BoxQP in blue. The convergence is identical to orange, which corresponds to changing the L2 to the Infinity norm for gradient tolerance checks.
2. The new BoxQP in purple. Convergence is initially faster, but then becomes linear and is not as good as the original BoxQP.
3. The original BoxQP, but changed to an exponential line search. Clearly, the performance is worse. This may be related to the the bug you identified but without further detail on where it was / how it manifested it's really hard to evaluate what's going on.

Any insight is highly appreciated

wxmerkt, 2020-04-02:
I found one bug in the old active-set for the special case that it converged during the first iteration. In this case `free_idx` was not set correctly, which lead to the feedback matrix `K` being incorrect. However, it does not have an effect on convergence (coincides with the orange line).

cmastalli, 2020-04-02:
@wxmerkt did you check that all the cases use the same regularization in each iteration?

It looks `OLD - X` cases are starting closer to the steepest descent direction. Note that the new Box-QP uses a very small default regularization as default.

wxmerkt, 2020-04-02:
Regularization is fixed and the same (1e-5) in all cases.

wxmerkt, 2020-04-02:
I have been digging deeper:

- The regularization is now passed in from DDP and adapted.
- The old-BoxQP active set bug was fixed (one line, as above).
- The timing is the same for old and new BoxQP (within 0.5 1e-8). We also tested them with random problems against SciPy's TNC and L-BFGS-B.

[image]

This is the result for a DDP problem with a quadcopter. The problem is exactly the same and we can switch the BoxQP with a config parameter. I don't quite understand why the old BoxQP with the fixed active set performs so much better.

cmastalli, 2020-04-02:
Are you using the same BoxDDP algorithm when you generate these results?

wxmerkt, 2020-04-03:
Yes, it's the same BoxDDP. I have exposed the ability to switch the underlying BoxQP with a configuration flag and ran several tests (different line search, different BoxQP, etc.). I understand the numerical behaviour of both very well now. 

The only difference I cannot grasp is: All else being equal, swapping Cholesky/LLT for the general inverse changes the convergence behaviour of the BoxDDP significantly. Hence, I am computing both and added a test to compare the two with `Hff_inv_cholesky.isApprox(Hff_inv_inverse, 1e-12)`: The two inverses match, but the behaviour is still different (Ubuntu 18.04, stock Eigen).

jcarpent, 2020-04-03:
@wxmerkt Which one is giving you better convergence rate?

wxmerkt, 2020-04-03:
The generalized inverse (I think it uses `PivLU` under the hood). Note, we are checking the `ComputationInfo` of the LLT and would throw an exception if it failed. And the inverse matrices appear to be the same.

I am currently working on a new plot looking into these different changes more closely.

cmastalli, 2020-04-03:
I wonder if the Cholesky decomposition creates numerically very small unsymmetric in `Hff_inv`. 

Could you try to impose symmetric? or check it?
For the former, you need to modify [118 line](https://github.com/loco-3d/crocoddyl/blob/devel/src/core/solvers/box-qp.cpp#L118) as follows:
```cpp
Hff_inv_llt_.solveInPlace(solution_.Hff_inv);
solution_.Hff_inv = 0.5 * (solution_.Hff_inv + solution_.Hff_inv.transpose()).eval()
```

jcarpent, 2020-04-03:
@wxmerkt You should try with LDLt and LLT. It is much more stable.

wxmerkt, 2020-04-03:
Thank you for the suggestions - I tested both:

1. Enforcing symmetry has no effect on the convergence of the BoxDDP.
2. Using the PivLU appears to lead to better convergence than LLT. I did not test LDLT. [I usually always go straight for LDLT or LLT and not the regular inverse, so am quite surprised and confused by this stark difference]

jcarpent, 2020-04-03:
Could you try LDLt, for sure it will be more stable at the numerical level.

cmastalli, 2020-04-08:
@wxmerkt do you have any update regarding this topic?

When it could be possible to try out the improvement that you did?

wxmerkt, 2020-04-09:
My laptop motherboard died last night - and it is unclear whether there is on-site support during Easter or the lockdown at all. I will hear more from IT next week after the Easter shutdown.
In parallel, I am trying to get access to the data on the laptop but waiting for parts from Amazon for that... I'll keep you posted.

cmastalli, 2020-04-09:
> My laptop motherboard died last night - and it is unclear whether there is on-site support during Easter or the lockdown at all. I will hear more from IT next week after the Easter shutdown.
> In [...]

Oh bad thing for this period.
It could work if you try to push the code, I could finish if this the case.

cmastalli, 2020-04-30:
In the box-QP, as well as in the DDP, we are using a LLT Cholesky decomposition. I have written a very simple code to bench it:
```cpp
#include <iostream>
#include <chrono>
#include <Eigen/Core>
#include <Eigen/Cholesky>
using namespace std;
using namespace std::chrono;
using namespace Eigen;

int main()
{
    unsigned int T = 1e6; // number of trials
    int N = 36; // 36 // matrix dimension
    MatrixXf cov = MatrixXf::Random(N, N);
    cov = (cov + cov.transpose()) + 1000 * MatrixXf::Identity(N, N);
    VectorXf b = VectorXf::Random(N), r1, r2;

    r1 = b;
    LLT<MatrixXf> llt;
    auto start = high_resolution_clock::now();
    for (unsigned int i = 0; i < T; ++i) {
        llt.compute(cov);
    }
    if (llt.info() != Success)
    {
        cout << "Error on LLT!" << endl;
        return 1;
    }
    auto middle = high_resolution_clock::now();
    for (unsigned int i = 0; i < T; ++i) {
        llt.solveInPlace(r1);
    }
    auto stop = high_resolution_clock::now();
    cout << "LLT decomposition & solving in  " << duration_cast<microseconds>(middle - start).count() / T
         << " + " << duration_cast<microseconds>(stop - middle).count() / T << " us." << endl;

    r2 = b;
    LDLT<MatrixXf> ldlt;
    start = high_resolution_clock::now();
    for (unsigned int i = 0; i < T; ++i) {
        ldlt.compute(cov);
    }
    if (ldlt.info() != Success)
    {
        cout << "Error on LDLT!" << endl;
        return 1;
    }
    middle = high_resolution_clock::now();
    for (unsigned int i = 0; i < T; ++i) {
        ldlt.solveInPlace(r2);
    }
    stop = high_resolution_clock::now();
    cout << "LDLT decomposition & solving in " << duration_cast<microseconds>(stop - start).count() / T
         << " + " << duration_cast<microseconds>(stop - middle).count() / T << " us." << endl;

    cout << "Total result difference: " << (r2 - r1).cwiseAbs().sum() << endl;
    return 0;
}
```

and the results for 76 and 36 matrix dimensions are
```bash
LLT decomposition & solving in  19 + 1 us.
LDLT decomposition & solving in 23 + 1 us.
Total result difference: 0
```
and 
```bash
LLT decomposition & solving in  3 + 0 us.
LDLT decomposition & solving in 4 + 0 us.
Total result difference: 0
```
respectively.

The question is "what do we prefer?". The stability of LDLT or the speed of the LLT solver.
We might also propose a solution were the default solver is LLT and the user can change to LDLT if this is they wish. To avoid unnecessary creation of data, we might do it through a template.

However, it would be great if @wxmerkt can provide a plot that shows the numerical behavior of both in the context of the DDP and Box-DDP algorithms.

jcarpent, 2020-04-30:
@cmastalli I think the way you perform the bench is not fair. Why are you using floats for your bench? you are always using the same matrix for the decomposition?
You are also doing online memory allocation when doing the first call.

You've printed two outputs. What is the second?

cmastalli, 2020-04-30:
> @cmastalli I think the way you perform the bench is not fair. Why are you using floats for your bench?

I agreed with you, but it is a fact that LLT is faster than LDLT. The explanation is that "the LLT implementation leverage cache-friendly matrix-matrix operations, while the LDLT implementation involves pivoting and matrix-vector operations only" [cit. https://stackoverflow.com/questions/23309596/eigen-ldlt-slower-than-llt]

> You've printed two outputs. What is the second?

Results for different matrix dimension (first, 76; second, 36).

jcarpent, 2020-04-30:
Yes, but in your case, you should advocate for numerical stabillity rather than performances (with only minor diff here).

jcarpent, 2020-04-30:
Can you redo the bench for double matrices too?

cmastalli, 2020-04-30:
> Yes, but in your case, you should advocate for numerical stabillity rather than performances (with only minor diff here).

To be honest, I run all the examples using LDLT approach in both QP and DDP algorithms. I didn't find a significant difference with the LDLT approach. For instance, the number of iterations and final cost values are the same, with an exception in the `bipedal_walk` where the LDLT requires around 8 iterations more than LLT.

My preliminary test shows no benefits of using LDLT, but I truly believe we need to investigate deeper before making a conclusion.

> Can you redo the bench for double matrices too?

These are the results with doubles and for 76 and 36 matrix dimension
```bash
LLT decomposition & solving in  25 + 1 us.
LDLT decomposition & solving in 31 + 1 us.
Total result difference: 0
```
,
```bash
LLT decomposition & solving in  4 + 0 us.
LDLT decomposition & solving in 5 + 0 us.
Total result difference: 0
```
respectively.

cmastalli, 2022-04-29:
@wxmerkt -- I noticed an error in the Armijo condition, which it is fixed in #1067.
This error was often running extra iteration even though that solution was found.

This could explain why the computation time is bigger in the Crocoddyl implementation, and it is not related with the factorisation solver.

Furthermore, this error was not guarantee convergence for random initialisation.

wxmerkt, 2022-05-02:
I'm currently on leave and will return to this on my return - I have some open changes for the BoxQP which also addressed the issue you likely solved in #1067. When I benchmarked those changes some time ago, it reduced computation time by ~90% for BoxQP/30% for Box(F)DDP. I'll circle back mid-next week

cmastalli, 2022-05-21:
I think we can close this issue, anyone in opposition to?

wxmerkt, 2022-05-24:
Sure - happy to close. I still have some commits related to it I'll sort out / PR after ICRA

closed 2022-05-24 by wxmerkt