loco-3d/crocoddyl #746: Questions about crcoddyl framework

TaeilJin, 2020-04-06:
Hi
I am installed crocoddyl on Linux, my code editor is a VScode and I use a python.

I am a beginner about optimal control and crocoddyl framework, so it might be a silly question.

I am trying to set a inequality constraints to constrain effector-position of quadruped have to be inside of convex hull.

Is there a way to adding new equality/inequality constraints? 

I also trying to see a c++ code of the function in crocoddyl for example crcoddyl.CostModelCoMPosition, But it is hard to access the function in editor.

Can I ask which is the debugging tools when you try to debug the code?

Thank you!

Taeil Jin

P.S.
I used a robotpkg to install the crocoddyl, and it had a issue that osgconv can't convert the Anymal.dae files. 
I solve this issue after I install the robotpkg "robotpkg-osg-dae" and "robotpkg-ur5-description" 
It is mentioned in pinnochio homepage, but not in crocoddyl github, so I think that it will be better to mention those robotpkg installation in Crocoddyl github page too.

cmastalli, 2020-04-06:
Hi @TaeilJin 

Thank you for being interested in Crocoddyl.
You have different questions, I will address them individually.

> I am a beginner about optimal control and crocoddyl framework, so it might be a silly question.
Before answering the questions. You need to consider that Crocoddyl solvers are based on differential dynamic programming (or better) the Bellman principle. They use the Markoviam assumption to implement dynamics, constraint and cost functions. You might want to spend some time reading our paper: https://cmastalli.github.io/publications/crocoddyl20icra.html

> I am trying to set a inequality constraints to constrain effector-position of quadruped have to be inside of convex hull.

Crocoddyl uses full dynamics, you do not need to define a box that satisfies kinematic feasibility (as commonly done in Centroidal dynamics trajectory optimization). If that your case, you just need to define state bounds.

> Is there a way to adding new equality/inequality constraints?

It is possible to do both: equality and inequality constraints.
The former could be done by extending the physical dynamic equation in the set of nodes that you want to define the equality constraint.
The latter could be introduce by adding quadratic penalization cost functions but it requires a high knowledge of optimization. We are working in a solver that will handle arbitrary inequality constraints, but at the moment it is ongoing research.

> I also trying to see a c++ code of the function in crocoddyl for example crcoddyl.CostModelCoMPosition, But it is hard to access the function in editor.

I am not sure what it is your question. Crocoddyl classes use a scalar template that are forwarded declare. I also use visual studio code, and I do not have any problem. You might need to setup the installation path.

> Can I ask which is the debugging tools when you try to debug the code?
Standard tools such as `gdb`. I suggested to you to start implementing things in Python. Crocoddyl allows us to prototype in both languages.

cmastalli, 2020-04-08:
I will close this issue since there is no any activity.

Feel free to re-open it if you need further help.

closed 2020-04-08 by cmastalli