loco-3d/crocoddyl #747: Questions about the framework of crocoddyl

ddliugit, 2020-04-08:
Hi, 

I noticed you updated the library with most c++ code. I am using API to construct an optimization problem and wondering if I can write new cost function for Pinocchio-based model with Python.
Can you show me some examples?

cmastalli, 2020-04-08:
Yes, it is possible to inheritance `CostModelAbstract` in Python. You only need to consider that it is not possible to derived new cost data structures (from `CostDataAbstract`), which in practice means that you have to create internal data inside the model class.

For more details see this example:
https://github.com/loco-3d/crocoddyl/blob/devel/bindings/python/crocoddyl/utils/__init__.py#L439

cmastalli, 2020-04-08:
My previous reply answers your question.
I will close this issue, feel free to re-open it.

closed 2020-04-08 by cmastalli