loco-3d/crocoddyl #749: is crocoddyl a library for iLQR based solver or DDP based solver?

ddliugit, 2020-04-08:
It seems this solver only used the first derivatives of the dynamics instead of second derivatives which should be included in DDP.

cmastalli, 2020-04-08:
We use only the Jacobian of the dynamics. It is equivalent to an iterative LQR algorithm.

Despite that, we prefer to think that our solver it is a DDP algorithm with Gauss-Newton assumption.

cmastalli, 2020-04-08:
Crocoddyl has different solvers. One of our novel solver is called feasibility-driven DDP which it is a remarkable improved version of the DDP algorithm.

Additionally, our DDP algorithm is not exactly the classical implementation. We make a relinearization of the Value function to be able to accept the state trajectory as warm-start.

For more details see the Crocoddyl paper described in Readme file.

cmastalli, 2020-04-19:
I will close this issue.

Feel free to ask further questions in other issues.

closed 2020-04-19 by cmastalli