loco-3d/crocoddyl #752: AttributeError: 'CostModelContactFrictionCone' object has no attribute 'frame'

julesser, 2020-04-10:
Hey guys, 
It seems that  some change in v1.2.0 caused `python3 bipedal_walk.py display` to fail.
**Output**: 
```
  File "bipedal_walk.py", line 79, in <module>
    ddp[i].solve(xs, us, 100, False, 0.1)
  File "/opt/openrobots/lib/python3.6/site-packages/crocoddyl/__init__.py", line 271, in __call__
    self.visualization.displayFromSolver(solver)
  File "/opt/openrobots/lib/python3.6/site-packages/crocoddyl/__init__.py", line 115, in displayFromSolver
    fs = self.getForceTrajectoryFromSolver(solver)
  File "/opt/openrobots/lib/python3.6/site-packages/crocoddyl/__init__.py", line 176, in getForceTrajectoryFromSolver
    if contact.joint == self.robot.model.frames[c.cost.frame].parent:
AttributeError: 'CostModelContactFrictionCone' object has no attribute 'frame'
```
Hopefully this report is useful to you :)

cmastalli, 2020-04-10:
Hi @julesser,

This issue has fixed in v1.2.1 but we are still waiting for the release.
Alternatively, you could install Crocoddyl by source.

I will close this issue since it has been fixed.

Carlos

julesser, 2020-04-10:
Okay, great.

closed 2020-04-10 by julesser

julesser, 2020-04-14:
FYI: Updating robotpkg-py36-crocoddyl to version 1.2.1 has fixed the problem for me.