loco-3d/crocoddyl #756: Inquiry about debugging binding c++ code

TaeilJin, 2020-04-17:
Hi
I am always thanks for your quick responses. 
I have a another question, Is there a way to debug binded c++ code?

I want to debug a function, for example, I want to debug the steps inside of crocoddyl.CostModelCoMPosition which is written by c++ code and binded

cmastalli, 2020-04-17:
You need to install Crocoddyl in DEBUG mode, and then you could use `gdb` (including with a Python script).

cmastalli, 2020-04-17:
But note that we have unittest code for checking the behavior of our functions. You could run it with
```bash
make test
```

TaeilJin, 2020-04-17:
Thank you for quick response!, 

To be honest, I was just following installation procedure (using robotpkg, or build from source). 
Is there a way to change a "mode"?

cmastalli, 2020-04-17:
Using normal cmake compilation, i.e.
```bash
mkdir build && cd build
cmake -DCMAKE_BUILD_TYPE=Debug ../
or
cmake ../ & make edit_cache
```

For robotpkg, you're installing Crocoddyl through a pre-compile binary. If you did it, you will need to first uninstall it.

TaeilJin, 2020-04-17:
Thank you so much!

cmastalli, 2020-04-17:
You're welcome

closed 2020-04-17 by cmastalli