loco-3d/crocoddyl #762: Casting an ActionModel from Scalar to a NewScalar type

proyan, 2020-04-22:
This issue is concerned with easy transcription and use of the templated classes in crocoddyl.

We currently need a virtual member function in all classes that can cast the class from one scalar type to another. This would be useful in codegenerating actionmodels. The typical use would be to cast the actionmodel to the CodeGen Scalar Type. So instead of Lines 145-245 in https://github.com/proyan/crocoddyl/blob/topic/codegen2/benchmark/bipedal-with-contact-cg.cpp, we would have the following two lines:

```
ad_runningModel = runningModel.cast<ADScalar>
ad_terminalModel = terminalModel.cast<ADScalar>
```

Unfortunately, there are two problems that we face:
1) All member functions are private, and we don't have accessors for _each and every_ member function of a class. These need to be created.
2) The `cast` function has to be virtual, and thus cannot be templatized. What this means is we need  to define a kind of visitor mechanism which visits `cast` for a fixed number of Scalar types, which would be predefined somewhere in the code. See https://stackoverflow.com/a/5872633 for full explanation.

There are three phases for this task, and they would follow the following order:
a) The point (1) is quite boring, and time consuming. 
b) Setting up the visitor mechanism is a bit technical, and is a bit rewarding.
c) Defining the casting function for all classes, which is again boring work.

cmastalli, 2020-04-24:
Let's do an exercise since it requires a lot modifications.
Could you write a simple example of how you would do it in the action model class with a simple c++ derivation?

Additionally, let's break this milestone in the three main PRs: one per point that you have mentioned.

cmastalli, 2020-07-24:
I am planning to start to work in this task. However, before that, I would like to propose a modification of this request.

Instead of creating a `cast<scalar>` function, I propose to do `toCoGen` function. My points are:
 - we don't want to support other scalars (already with this single scalar there is a lot of to be written for each single class in Crocoddyl)
 - there is no need for casting scalars like float or CasADi

Rhis suggestion limits the use to other cases, but I truly believe we don't want to support so many things (we don't have the resources neither motivation).

@proyan and @nmansard. I need your support on this decision for starting to work. I am also happy to do not do it, but I don't think we have the resources

proyan, 2020-07-26:
Hi Carlos,

Don't bother, I had already started implementing it. If you need it, I can try to finish it as a priority, but I don't see why you would need this immediately?

cmastalli, 2020-07-27:
Thanks for taking care. Yes, it is important to me because I want to start with its Python bindings. The objective is to try it within my MPC algorithm written in Python. I am trying to avoid to write it in c++, so I could easily develop in future more advance ideas.