loco-3d/crocoddyl #763: Action or Cost model for following desired trajectories

TaeilJin, 2020-04-27:
Hi

I plan to optimize the CoM and foot trajectories, and than create the whole quadruped motion that follows the trajectories which I have created.

Looking at the example, it seems to create their own trajectories from the user input. (ex: number of knots, step height and length ..) 

- Can you tell me which functions I should refer to ?

- If you can, I want to know which is the cost function and constraints for quadrupedal_gaits.py.

cmastalli, 2020-04-27:
You could definitely formulate an optimal control problem that tracks a desired centroidal dynamics trajectory (i.e. CoM position, angular momentum regulation, and force tracking). And you are right, our examples are written to track a set of foothold locations.

Unfortunately, at the moment, we don't have any example of how to formulate these kind of problem. However, you could explore the different cost functions that we have.

The implementation of the quadrupedal example is experimental, and I am currently working in a much better one. You could read the code inside here: https://github.com/loco-3d/crocoddyl/blob/devel/bindings/python/crocoddyl/utils/quadruped.py.

Let me know if this helped you.

We are planning to write a set of Jupyter Notebooks for learning each aspect of the library.

cmastalli, 2020-05-08:
I will close this issue since there is no activity or actual bug or feature to be implemented

closed 2020-05-08 by cmastalli